project-chip/connectedhomeip #26303: [chiptool.py] `chiptool.py tests` command missing support for several command line arguments used by Matter TH

mikaelhm, 2023-04-28:
### Reproduction steps / Feature

**Steps to reproduce:**
1. Build`chip-tool and `chip-all-clusters-app` in SDK Repo
2. Run `chip-all-clusters-app`
3. Perform "paring" using `chiptool.py` from same ethernet
    
    ```sh
    ./scripts/tests/yaml/chiptool.py pairing onnetwork-long 0x12344321 20202021 3840
    ```
    
4. Attempt to run a test case Matter TH:
    
    ```sh
    ./scripts/tests/yaml/chiptool.py tests Test_TC_ACE_1_1 \
       --nodeId 0x12344321 \
       --delayInMs 250 \
       --timeout 900 \
       --continueOnFailure 1 \
       --trace_file "/tmp/trace_log_TEST_Test_TC_ACE_1_1.log" \
       --trace_decode 1
    ```

I tried removing the parameters one at a time, and verified that the following arguments are not supported:
- `--delayInMs`
- `--timeout`
- `--continueOnFailure`
- `--trace_file`
- `--trace_decode`

**Note:** `--nodeId` is supported, but never works as it raises a `TypeError` no matter the format of the value:
```
"/root/examples/chip-tool/py_matter_chip_tool_adapter/matter_chip_tool_adapter/encoder.py", line 186, in __maybe_add_destination
    destination_argument_value = hex(request.node_id)
TypeError: 'str' object cannot be interpreted as an integer
```

### Platform

python

### Type

Manually tested with SDK

### Anything else?

`continueOnFailure` and `delayInMs` commands are implemented here for `chip-tool` binary:
https://github.com/project-chip/connectedhomeip/blob/77d72a84039fea5b551378d9f7d5119f5d9a16ab/examples/chip-tool/commands/tests/TestCommand.h#L52C1-L54

`timeout` is defined here for `chip-tool` binary:
https://github.com/project-chip/connectedhomeip/blob/77d72a84039fea5b551378d9f7d5119f5d9a16ab/examples/chip-tool/commands/clusters/ModelCommand.h#LL52C1-L52C1

bzbarsky-apple, 2023-04-28:
@vivien-apple

krypton36, 2023-04-28:
When adding these, can you please make it where it is part of some config in the runner? chiptool.py can update the parameters of the config base on the command line args. The reason I ask for this is because I am working on implementing a new runner specifically for the test harness and we will need these arguments to be available directly as part of a config.

PR #26333 (2023-05-05): [chiptool.py] Add --trace_file, --trace_decode, --node_id, --continue…
…OnFailure, --timeout, --delayInMs supports

#### Problem

Those arguments are not parsed correctly by `chiptool.py` at the moment, which prevents the test harness to be updated.

Fix #26303
closed 2023-05-05 by vivien-apple