project-chip/connectedhomeip #26313: [BUG] [chip-tool] cannot generate QR CODE

mmt-10, 2023-05-01:
### Reproduction steps

```bash
$ ./chip-tool payload generate-qrcode --discriminator 3840 --setup-pin-code 20202021 --vendor-id 65521 --product-id 32768 --version 1 --commissioning-mode 0
```

> [1682942421.206178][122331:122331] CHIP:TOO: Run command failure: src/setup_payload/QRCodeSetupPayloadGenerator.cpp:268: Error 0x0000002F

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 309e9f1afc989cea5927063d8a592e5eecb0b3ea (HEAD -> master, origin/master, origin/HEAD)

### Platform

other

bzbarsky-apple, 2023-05-01:
>  As my understanding, --version means firmware version for application.

No, `--version` is the version of the encoding used to represent Matter data in the byte string that is then represented by the QR code (not to be confused with the QR code version or with any sort of other Matter version).

See "Packed Binary Data Structure" in the spec, which says for the Version field:

> 3-bit value specifying the QR code payload version.
> SHALL be 000.

So a nonzero version is not a valid QR code for Matter so far; this value is present as an extension point in case we ever need to change the way Matter QR codes are encoded.

Since nonzero is not a valid value, and `--allow-invalid-payload` was not passed to the `generate-qrcode` command, the generation failed.  The error message could be a bit more helpful in this case....

closed 2023-05-01 by bzbarsky-apple

mmt-10, 2023-05-02:
Thank you for your reply.
I understand it.
I will check specification.

BTW, would you please add "guide for `generate-qrcode`" to [https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md)?

bzbarsky-apple, 2023-05-02:
@mmt-10 I can try to find time to do that, but you should feel free to do that yourself; just create a PR to that file...