project-chip/connectedhomeip #26318: [BUG] Chip-tool cannot connect to matter device, crashes after some time

marcelgoh, 2023-05-01:
### Reproduction steps

I have an ESP32 device with the `examples/light` example from `esp-matter` connected to the wifi network via `idf.py monitor`. From there I can commission and connect via the ESP Matter app, but if I try to connect from the CHIP tool, it hangs and then eventually quits with a critical error. Here is the output (after running `chip-tool pairing code-wifi 0x7283 <ssid> <password> 34970112332`):

<details>

<summary>Logs</summary>

```
[1682968103.753160][30473:30473] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1682968103.753432][30473:30473] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1682968103.753563][30473:30473] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1682968103.753656][30473:30473] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1682968103.753847][30473:30473] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-DQPbgq)
[1682968103.754190][30473:30473] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1682968103.754294][30473:30473] CHIP:DL: NVS set: chip-counters/reboot-count = 31 (0x1F)
[1682968103.754660][30473:30473] CHIP:DL: Got Ethernet interface: eth0
[1682968103.754824][30473:30473] CHIP:DL: Found the primary Ethernet interface:eth0
[1682968103.755032][30473:30473] CHIP:DL: Failed to get WiFi interface
[1682968103.755082][30473:30473] CHIP:DL: Failed to reset WiFi statistic counts
[1682968103.755123][30473:30473] CHIP:IN: UDP::Init bind&listen port=0
[1682968103.755240][30473:30473] CHIP:IN: UDP::Init bound to port=45250
[1682968103.755289][30473:30473] CHIP:IN: UDP::Init bind&listen port=0
[1682968103.755412][30473:30473] CHIP:IN: UDP::Init bound to port=40937
[1682968103.755449][30473:30473] CHIP:IN: BLEBase::Init - setting/overriding transport
[1682968103.755456][30473:30473] CHIP:IN: TransportMgr initialized
[1682968103.755492][30473:30473] CHIP:FP: Initializing FabricTable from persistent storage
[1682968103.755588][30473:30473] CHIP:TS: Last Known Good Time: 2023-05-01T12:07:15
[1682968103.755817][30473:30473] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xAA240AF72C9494DA, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1682968103.756773][30473:30473] CHIP:ZCL: Using ZAP configuration...
[1682968103.757546][30473:30473] CHIP:CTL: System State Initialized...
[1682968103.757609][30473:30473] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682968103.757627][30473:30473] CHIP:CTL: Setting attestation nonce to random value
[1682968103.757684][30473:30473] CHIP:CTL: Setting CSR nonce to random value
[1682968103.758949][30473:30475] CHIP:DL: CHIP task running
[1682968103.759053][30473:30475] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1682968103.759323][30473:30475] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682968103.759410][30473:30475] CHIP:CTL: Setting attestation nonce to random value
[1682968103.759467][30473:30475] CHIP:CTL: Setting CSR nonce to random value
[1682968103.761223][30473:30475] CHIP:CTL: Generating NOC
[1682968103.761727][30473:30475] CHIP:FP: Validating NOC chain
[1682968103.762386][30473:30475] CHIP:FP: NOC chain validation successful
[1682968103.762492][30473:30475] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1682968103.762505][30473:30475] CHIP:TS: Last Known Good Time: 2023-05-01T12:07:15
[1682968103.762562][30473:30475] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1682968103.762587][30473:30475] CHIP:TS: Retaining current Last Known Good Time
[1682968103.763763][30473:30475] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1682968103.764848][30473:30475] CHIP:TS: Committing Last Known Good Time to storage: 2023-05-01T12:07:15
[1682968103.765692][30473:30475] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: AA240AF72C9494DA)
[1682968103.771138][30473:30475] CHIP:CTL: Setting wifi credentials from parameters
[1682968103.771174][30473:30475] CHIP:CTL: Setting attestation nonce to random value
[1682968103.771205][30473:30475] CHIP:CTL: Setting CSR nonce to random value
[1682968103.771239][30473:30475] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682968103.771289][30473:30475] CHIP:CTL: Starting commissioning discovery over BLE
[1682968103.771343][30473:30475] CHIP:CTL: Starting commissioning discovery over DNS-SD
[1682968103.772720][30473:30474] CHIP:DL: FAIL: get bus sync in StartupEndpointBindings, error: Could not connect: No such file or directory
[1682968103.772856][30473:30475] CHIP:DL: PlatformBlueZInit init success
[1682968103.772877][30473:30475] CHIP:CTL: Commissioning discovery over BLE failed: ../../src/platform/Linux/BLEManagerImpl.cpp:682: CHIP Error 0x00000003: Incorrect state
[1682968103.772957][30473:30475] CHIP:-: ../../src/platform/Linux/BLEManagerImpl.cpp:682: CHIP Error 0x00000003: Incorrect state at ../../src/controller/SetUpCodePairer.cpp:299
[1682968103.772986][30473:30475] CHIP:BLE: No adapter available for new connection establishment
[1682968133.786533][30473:30475] CHIP:CTL: Discovery timed out
[1682968133.786737][30473:30475] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682968133.786769][30473:30475] CHIP:TOO: Secure Pairing Failed
[1682968133.788015][30473:30473] CHIP:CTL: Shutting down the commissioner
[1682968133.788184][30473:30473] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682968133.788212][30473:30473] CHIP:CTL: Shutting down the controller
[1682968133.788234][30473:30473] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1682968133.788253][30473:30473] CHIP:FP: Forgetting fabric 0x1
[1682968133.788560][30473:30473] CHIP:TS: Pending Last Known Good Time: 2023-05-01T12:07:15
[1682968133.788894][30473:30473] CHIP:TS: Previous Last Known Good Time: 2023-05-01T12:07:15
[1682968133.789002][30473:30473] CHIP:TS: Reverted Last Known Good Time to previous value
[1682968133.789070][30473:30473] CHIP:CTL: Shutting down the commissioner
[1682968133.789163][30473:30473] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1682968133.789256][30473:30473] CHIP:CTL: Shutting down the controller
[1682968133.789355][30473:30473] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1682968133.789599][30473:30473] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1682968133.789737][30473:30473] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1682968133.789775][30473:30473] CHIP:FP: Shutting down FabricTable
[1682968133.789869][30473:30473] CHIP:TS: Pending Last Known Good Time: 2023-05-01T12:07:15
[1682968133.790053][30473:30473] CHIP:TS: Previous Last Known Good Time: 2023-05-01T12:07:15
[1682968133.790147][30473:30473] CHIP:TS: Reverted Last Known Good Time to previous value
[1682968133.790364][30473:30473] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-sR7BLr)
[1682968133.790856][30473:30473] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1682968133.790972][30473:30473] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1682968133.790998][30473:30473] CHIP:DL: Inet Layer shutdown
[1682968133.791015][30473:30473] CHIP:DL: BLE shutdown

(process:30473): GLib-GObject-CRITICAL **: 13:08:53.791: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[1682968133.791368][30473:30473] CHIP:DL: System Layer shutdown
[1682968133.791912][30473:30473] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/pairing/PairingCommand.cpp:240: CHIP Error 0x00000003: Incorrect state
```

</details>

Is this the result of user error on my part, or a bona fide bug? Let me know if there are any more details I should supply.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

0b1762e5119c6cd4703ebb2166441187bff3fd9c

### Platform

windows

bzbarsky-apple, 2023-05-02:
The problem seems to be right here:
```
[1682968103.772720][30473:30474] CHIP:DL: FAIL: get bus sync in StartupEndpointBindings, error: Could not connect: No such file or directory
```
which is due to  this call failing:
```
    conn = g_bus_get_sync(G_BUS_TYPE_SYSTEM, nullptr, &error);
```

@marcelgoh That part seems like an issue with gdbus or bluez setup on your system, at first glance.  But it's _possible_ that you might need to `sudo` to connect to the system bus, depending on the gdbus version.

This part:
```
(process:30473): GLib-GObject-CRITICAL **: 13:08:53.791: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
```
looks like a real bug in the bluez bits; if init fails, it's trying to clean up something that never got set up.  Would be useful to know which exact unref this is coming from....  Is this from the:
```
    g_object_unref(apEndpoint->mpAdapter);
```
in `ShutdownBluezBleLayer` in `src/platform/Linux/bluez/Helper.cpp`?

jacqueshsu, 2023-08-24:
@bzbarsky-apple 
Do you know which the dbus version that can support system bus connection without sudo?

bzbarsky-apple, 2023-08-28:
@jacqueshsu I don't know, sorry.  You need someone more familiar with the Linux bits.

andy31415, 2024-10-24:
Platform seems to be Windows 10 with WSL ... while it maybe work, SDK maintainers do not currently have a windows maintainer.

andy31415, 2024-11-05:
I will have to close this one: very old bug and we do not generally have WSL support. We only really can debug Ubuntu as this is what we use for our docker images (and maybe some other linux distro on a case by case basis depending on what the main maintainers run on their system)

closed 2024-11-05 by andy31415