project-chip/connectedhomeip #26345: [1.0]

SYESHASWINI, 2023-05-03:
### Reproduction steps

When I try to perform OTA using chip-tool, the very first time I get the error that **esp ota abort failed** and the ota stops. With the second attempt the ota process gets completed and the device is restarts successfully.

After the restart of device  it shows like the failed to confirm image 3. what does this mean , is this any kind of error.
**E (2752) chip[SWU]: Failed to confirm image: 3**

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

none

### Platform

esp32

### Type

Core SDK Performance Improvement

bzbarsky-apple, 2023-05-03:
@SYESHASWINI What exact commands did you run?   Please attach (not paste) logs for the relevant devices.  It's really hard to say anything about your situation otherwise.

>  E (2752) chip[SWU]: Failed to confirm image: 3

That's what you would get if the running version does not match the target version after a restart.

SYESHASWINI, 2023-05-04:
@bzbarsky-apple attaching the error logs of confirm ota image failed.

[Ota_rquestor1.txt](https://github.com/project-chip/connectedhomeip/files/11394280/Ota_rquestor1.txt)
[ota_requestor2.txt](https://github.com/project-chip/connectedhomeip/files/11394281/ota_requestor2.txt)
[chip-tool log.txt](https://github.com/project-chip/connectedhomeip/files/11394282/chip-tool.log.txt)

SYESHASWINI, 2023-05-04:
@bzbarsky-apple this issue is regarding the **ESP abort failed error,** which across for very first time of performing ota and performing the same process second time which result of successfully ota with version match fail.

is there any way to avoid the esp abort failed error as this comes after completely downloading the ota image from requestor side.

closed 2023-06-13 by SYESHASWINI