project-chip/connectedhomeip #26351: [1.0]

suryaelvino, 2023-05-03:
### Reproduction steps

Step to build project

1. Source scripts/activate.sh
2. Source scripts/bootstrap.sh
3. gn gen out/host
4. ninja -C out/host
5. idf.py build project

### Bug prevalence

When i running to build project

### GitHub hash of the SDK that was being used

1

### Platform

esp32, python

### Platform Version(s)

esp32 dev kit

### Type

Platform Issue, Core SDK Crash

### Anything else?

```
[1121/1126] Generating ld/sections.ld
warning: the default selection DEVICE_TYPE_ESP32_H2_DEVKITC (undefined) of <choice> (defined at /home/suryaelvino/connectedhomeip/examples/lighting-app/esp32/main/Kconfig.projbuild:23) is not contained in the choice
[1122/1126] Building C object CMakeFiles/chip-lighting-app.elf.dir/project_elf_src_esp32.c.obj
[1123/1126] Linking CXX executable chip-lighting-app.elf
FAILED: chip-lighting-app.elf 
: && /home/suryaelvino/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  -g  CMakeFiles/chip-lighting-app.elf.dir/project_elf_src_esp32.c.obj -o chip-lighting-app.elf  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/espressif__mdns/libespressif__mdns.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  esp-idf/chip/libchip.a  esp-idf/QRCode/libQRCode.a  esp-idf/espressif__led_strip/libespressif__led_strip.a  esp-idf/main/libmain.a  esp-idf/esp32_mbedtls/libesp32_mbedtls.a  esp-idf/spidriver/libspidriver.a  esp-idf/tft/libtft.a  esp-idf/screen-framework/libscreen-framework.a  -Wl,--cref  -Wl,--defsym=IDF_TARGET_ESP32=0  -Wl,--Map="/home/suryaelvino/connectedhomeip/examples/lighting-app/esp32/build/chip-lighting-app.map"  -fno-lto  -Wl,--gc-sections  -Wl,--warn-common  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/unity/libunity.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/QRCode/libQRCode.a  esp-idf/espressif__led_strip/libespressif__led_strip.a  esp-idf/chip/libchip.a  esp-idf/bt/libbt.a  -L/home/suryaelvino/esp/esp-idf/components/bt/controller/lib_esp32/esp32  -lbtdm_app  -u ld_include_hli_vectors_bt  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/espressif__mdns/libespressif__mdns.a  esp-idf/console/libconsole.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  -Wl,--start-group  esp-idf/chip/lib/libCHIP.a  esp-idf/bt/libbt.a  -lbtdm_app  esp-idf/console/libconsole.a  esp-idf/main/libmain.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/esp32_mbedtls/libesp32_mbedtls.a  -Wl,--end-group  esp-idf/tft/libtft.a  esp-idf/spidriver/libspidriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libcore.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libespnow.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libmesh.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libnet80211.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libsmartconfig.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libwapi.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi/lib/esp32/libcore.a  /home/suryaelvino/esp/esp-idf/components/esp_wifi [... cut ...] data._ZTI21CommonDeviceCallbacks[_ZTI21CommonDeviceCallbacks]+0x8): undefined reference to `_ZTIN4chip13DeviceManager26CHIPDeviceManagerCallbacksE'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/suryaelvino/connectedhomeip/examples/lighting-app/esp32/build/log/idf_py_stderr_output_43502 and /home/suryaelvino/connectedhomeip/examples/lighting-app/esp32/build/log/idf_py_stdout_output_43502
```

bzbarsky-apple, 2023-05-03:
@dhrishi @shubhamdp

shubhamdp, 2023-05-04:
@suryaelvino What example are you trying to build?
To build an ESP32 example please find the documentation guide from here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/README.md

shubhamdp, 2023-05-10:
@suryaelvino  any update here?

dhrishi, 2024-10-25:
Closing due to no update. Please reopen in case of issues

closed 2024-10-25 by dhrishi