project-chip/connectedhomeip #26353: [1.1.0.1] venv activation/bootstrap error

raven-worx, 2023-05-03:
### Reproduction steps

Hi,

after a few months i decided to upgrade the SDK checkout to yesterdays tag "v1.1.0.1" for my arm crosscompile dynamic bridge app.

During activation/bootstrapping of the venv pigweed/CIPD runs into the following error. (So this error occurs before i even configure for an arm build in any way.)

> Errors:
  failed to install fuchsia/third_party/armgcc/linux-amd64 in "arm": adding files to the site root: lstat /mnt/c/dev/matter/connectedhomeip/examples/my-bridge/third_party/connectedhomeip/.environment/cipd/packages/.cipd/pkgs/0/9z1FnQUHn-LGwcPNIsxnZzTE3HjoopYzwWlK3f1-HG0C/share/doc/gdb/qMemTags.html: no such file or directory
/mnt/c/dev/matter/connectedhomeip/examples/my-bridge/third_party/connectedhomeip/.environment/cipd/cipd ensure -ensure-file /mnt/c/dev/matter/connectedhomeip/examples/my-bridge/third_party/connectedhomeip/.environment/cipd/packages.ensure -root /mnt/c/dev/matter/connectedhomeip/examples/my-bridge/third_party/connectedhomeip/.environment/cipd/packages -log-level debug -json-output /mnt/c/dev/matter/connectedhomeip/examples/my-bridge/third_party/connectedhomeip/.environment/cipd/packages.json -cache-dir /home/me/.cipd-cache-dir -max-threads 0

what might be the cause?

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

8f66f4215bc0708efc8cc73bda80620e67d8955f

### Platform

other

andy31415, 2023-05-03:
What os/env is this? Arm seems to indicate an arm cpu?

raven-worx, 2023-05-03:
@andy31415 
i am actually building inside WSL (Ubuntu) - which was fine so far
the error also comes up even before the call to "gn gen ..." to configure my project for cross compilation. 
I also tried a complete new checkout without any leftovers,

andy31415, 2023-05-03:
The error seems to be in CIPD, logs seem to indicate `/mnt/c/dev/matter/connectedhomeip/examples/my-bridge/third_party/connectedhomeip/.environment/cipd/cipd ensure ...` however I am not sure of the full command.

The way things work is that cipid is supposed to fetch large packages from the internet and `ensure` makes sure that you have those things downloaded locally.

Did you try on a clean environment (i.e. remove the entire .environment and run bootstrap in a fresh login shell)?

raven-worx, 2023-05-04:
i think i found the issue

after inspecting the armgcc package, i noticed that there are 2 files with the same name. One "QMemTags.html" and one "qMemTags.html" in the same path. Since my checkout is in the Windows filesystem mount this doesnt work out very well in the end :/

I think i have to try to move the .environment folder into the "native" WSL FS via symlink or so.

closed 2023-05-04 by raven-worx

bzbarsky-apple, 2023-05-04:
@raven-worx Hmm...  That will fail on Mac too, presumably, by default.  Might be worth getting the package fixed, since I doubt it's intended to have both those files there.

sammichael25, 2023-07-22:
@raven-worx  did you ever get this resolved. I'm getting the same error.

raven-worx, 2023-07-22:
@sammichael25 
Yes, as i wrote 2 comments above. Ensure that you are in a case-sensitive filesystem

xuanya0, 2024-11-24:
ran into the same issue in a virtiofs mounted path on linux

turns out if you run `unzip` manually it's going to see the second file as already existing