project-chip/connectedhomeip #26356: [Platform] EFR32 lighting-app BRD2704A

lboue, 2023-05-03:
### Reproduction steps

Hello,

I am failing to build EFR32 lighting-app for BRD2704A. I may need some help.

### Reproduction steps

```
user@SL-Matter:~/matter/examples/lighting-app/silabs/efr32$ export SILABS_BOARD=BRD2704A
user@SL-Matter:~/matter/examples/lighting-app/silabs/efr32$ gn gen out/debug
Done. Made 2825 targets from 254 files in 230ms

user@SL-Matter:~/matter/examples/lighting-app/silabs/efr32$ ninja -C out/debug
ninja: Entering directory `out/debug'
[1/10] ld ./chip-efr32-lighting-example.out
FAILED: chip-efr32-lighting-example.out chip-efr32-lighting-example.out.map
arm-none-eabi-g++ -T../../third_party/connectedhomeip/examples/platform/silabs/efr32/ldscripts/mgm24.ld -march=armv8-m.main+dsp -mcpu=cortex-m33 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Og --specs=nosys.specs --specs=nano.specs -Werror -Wl,--fatal-warnings -fdiagnostics-color -Wl,--gc-sections -Wl,-Map,./chip-efr32-lighting-example.out.map @./chip-efr32-lighting-example.out.rsp -o ./chip-efr32-lighting-example.out
/home/user/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld:../../third_party/connectedhomeip/third_party/silabs/gecko_sdk/protocol/openthread/libs/libsl_openthread_efr32mg2x_gcc.a: file format not recognized; treating as linker script
/home/user/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld:../../third_party/connectedhomeip/third_party/silabs/gecko_sdk/protocol/openthread/libs/libsl_openthread_efr32mg2x_gcc.a:1: syntax error
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Do you have any idea where this error could be coming from?

### Platform

efr32

### Type

Manually tested with SDK

### Anything else?

Commit 3e9f43e76df3a058f343a9dbc6ce4135c41439d7

jmartinez-silabs, 2023-05-04:
Same vein as the other bug you opened. I'll reproduce and provide a fix. There seems to be a linking error

lboue, 2023-05-04:
> Same vein as the other bug you opened. I'll reproduce and provide a fix. There seems to be a linking error

Do you know why this issue was not detect with CICD tests?

jmartinez-silabs, 2023-05-04:
@lboue Due to CI time constraints, not all boards we support are built in the github CI

closed 2023-05-04 by bzbarsky-apple

lboue, 2023-05-05:
Thanks for you help. 
Now build is build EFR32 lighting-app for BRD2704A is OK:
```
~/connectedhomeip/examples/lighting-app/silabs/efr32$ ls out/debug/
args.gn            chip-efr32-lighting-example.flash.py  chip-efr32-lighting-example.s37  lib                           __pycache__                         zap_gen.lock
build.ninja        chip-efr32-lighting-example.hex       efr32_firmware_utils.py          lighting_app.flashbundle.txt  python
build.ninja.d      chip-efr32-lighting-example.out       firmware_utils.py                obj                           relative_path_transformations.json
build.ninja.stamp  chip-efr32-lighting-example.out.map   gen                              protocol_buffer               toolchain.ninja
```

PR #26370 (2023-05-04): [Silabs] Fix linkage and some apps build for the Sparkfun board
fixes #26357
fixes #26356
fixes #26350

Two recently added files from Silabs SDK needed to be linked for that board.
This board has no buttons, so I added some `#ifdef` to the button's usage in certain apps.
Removed some include of the buttons header where they were not needed any more thanks to the ongoing silabs platforms abstraction