project-chip/connectedhomeip #26357: [BUG] EFR32 light-switch-app BRD2704A

lboue, 2023-05-03:
### Reproduction steps

### Reproduction steps

Hello,

I am failing to build EFR32 light-switch-app for BRD2704A. I may need some help.

### Reproduction steps

```
./scripts/examples/gn_efr32_example.sh ./examples/light-switch-app/silabs/efr32/ ./out/light-switch-app BRD2704A
....
[FAILED: [/src/chip-efr32-light-switch-example.out.AppTask.cpp.o 
arm-none-eabi-g++ -MMD -MF obj/src/chip-efr32-light-switch-example.out.AppTask.cpp.o.d -march=armv8-m.main+dsp -mcpu=cortex-m33 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Og -g2 -fno-common -ffunction-sections -fdata-sections --specs=nosys.specs --specs=nano.specs -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/util/plugin/security_manager/ -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/hardware/kit/common/bsp -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/app/common/util/app_assert/ -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/hardware/board/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/hardware/driver/memlcd/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/hardware/driver/memlcd/src/ls013b7dh03 -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/bootloader -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/bootloader/config -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/bootloader/config/s2/btl_interface -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/bootloader/api -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/CMSIS/Core/Include -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/CMSIS/RTOS2/Include -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/common/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/emdrv/common/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/emdrv/gpiointerrupt/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/emdrv/dmadrv/config -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/emdrv/dmadrv/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/emdrv/nvm3/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/emdrv/rtcdrv/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/emlib/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/halconfig/inc/hal-config -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/peripheral/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/radio/rail_lib/common -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/radio/rail_lib/chip/efr32 -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/radio/rail_lib/chip/efr32/rf/common/cortex -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/radio/rail_lib/protocol/ieee802154 -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/radio/rail_lib/plugin/pa-conversions -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/radio/rail_lib/plugin/rail_util_pti -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/radio/rail_lib/plugin/rail_util_rf_path -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/device_init/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/hfxo_manager/config/ -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/hfxo_manager/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/hfxo_manager/src -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/mpu/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/power_manager/config/ -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/power_manager/inc/ -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/power_manager/src/ -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/sleeptimer/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/sleeptimer/config -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/system/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/udelay/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/legacy_hal/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/token_manager/config -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/token_manager/inc -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/token_manager/test -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/token_manager/test/include -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/token_manager/test/stack/config -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/service/token_manager/test/stack/include -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/middleware/glib -isystem../../../examples/light-switch-app/silabs/efr32/third_party/connectedhomeip/third_party/silabs/gecko_sdk/platform/middleware/glib/config -isystem../../../examples/light-switch-app/silabs/efr32/third_party [... cut ...]
[01m[K../../../examples/light-switch-app/silabs/efr32/include/AppTask.h:32:10:[m[K [01;31m[Kfatal error: [m[Ksl_simple_button_instances.h: No such file or directory
   32 | #include [01;31m[K"sl_simple_button_instances.h"[m[K
      |          [01;31m[K^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~[m[K
compilation terminated.
```

Do you have any idea where this error could be coming from?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

3e9f43e76df3a058f343a9dbc6ce4135c41439d7

### Platform

efr32

### Anything else?

[EFR32 switch.log](https://github.com/project-chip/connectedhomeip/files/11391409/EFR32.switch.log)

lboue, 2023-05-03:
I noticed that _third_party/silabs/matter_support/matter/efr32/efr32mg24/BRD2704A/autogen/sl_simple_button_instances.h_ file is missing.
There is a reference to this file in [AppTask.h](https://github.com/project-chip/connectedhomeip/blob/7b60592f9390cdb93f402ac26c5b284848318c8e/examples/light-switch-app/silabs/efr32/include/AppTask.h#LL32C1-L32C1)

```
 cat third_party/silabs/matter_support/matter/efr32/efr32mg24/BRD2703A/autogen/sl_simple_button_instances.h
/***************************************************************************//**
 * @file
 * @brief Simple Button Driver Instances
 *******************************************************************************
 * # License
 * <b>Copyright 2019 Silicon Laboratories Inc. www.silabs.com</b>
 *******************************************************************************
 *
 * The licensor of this software is Silicon Laboratories Inc.  Your use of this
 * software is governed by the terms of Silicon Labs Master Software License
 * Agreement (MSLA) available at
 * www.silabs.com/about-us/legal/master-software-license-agreement.  This
 * software is distributed to you in Source Code format and is governed by the
 * sections of the MSLA applicable to Source Code.
 *
 ******************************************************************************/

#ifndef SL_SIMPLE_BUTTON_INSTANCES_H
#define SL_SIMPLE_BUTTON_INSTANCES_H

#include "sl_simple_button.h"

extern const sl_button_t sl_button_btn0;
extern const sl_button_t sl_button_btn1;

extern const sl_button_t *sl_simple_button_array[];

#define SL_SIMPLE_BUTTON_COUNT 2
#define SL_SIMPLE_BUTTON_INSTANCE(n) (sl_simple_button_array[n])

void sl_simple_button_init_instances(void);
void sl_simple_button_poll_instances(void);

#endif // SL_SIMPLE_BUTTON_INSTANCES_H
```

```
$ cat third_party/silabs/matter_support/matter/efr32/efr32mg24/BRD2704A/autogen/sl_simple_button_instances.h
cat: third_party/silabs/matter_support/matter/efr32/efr32mg24/BRD2704A/autogen/sl_simple_button_instances.h: No such file or directory
```

lboue, 2023-05-03:
### Workaround try 
```
cp -r third_party/silabs/matter_support/matter/efr32/efr32mg24/BRD2703A/ third_party/silabs/matter_support/matter/efr32/efr32mg24/BRD2704A
```
### Failure
```
$ cd ~/matter/examples/lighting-app/silabs/efr32
user@SL-Matter:~/matter/examples/lighting-app/silabs/efr32$ ninja -C out/debug
ninja: Entering directory `out/debug'
[1/10] ld ./chip-efr32-lighting-example.out
FAILED: chip-efr32-lighting-example.out chip-efr32-lighting-example.out.map
arm-none-eabi-g++ -T../../third_party/connectedhomeip/examples/platform/silabs/efr32/ldscripts/mgm24.ld -march=armv8-m.main+dsp -mcpu=cortex-m33 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Og --specs=nosys.specs --specs=nano.specs -Werror -Wl,--fatal-warnings -fdiagnostics-color -Wl,--gc-sections -Wl,-Map,./chip-efr32-lighting-example.out.map @./chip-efr32-lighting-example.out.rsp -o ./chip-efr32-lighting-example.out
/home/lboue/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld:../../third_party/connectedhomeip/third_party/silabs/gecko_sdk/protocol/openthread/libs/libsl_openthread_efr32mg2x_gcc.a: file format not recognized; treating as linker script
/home/lboue/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld:../../third_party/connectedhomeip/third_party/silabs/gecko_sdk/protocol/openthread/libs/libsl_openthread_efr32mg2x_gcc.a:1: syntax error
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

lboue, 2023-05-03:
Same issue for BRD4316A board

`SILABS_BOARD=BRD4316A`

```
user@SL-Matter:~/matter/examples/lighting-app/silabs/efr32$ export SILABS_BOARD=BRD4316A
user@SL-Matter:~/matter/examples/lighting-app/silabs/efr32$ gn gen out/debug
Done. Made 2826 targets from 255 files in 291ms
user@SL-Matter:~/matter/examples/lighting-app/silabs/efr32$ ninja -C out/debug
ninja: Entering directory `out/debug'
[856/863] ld ./chip-efr32-lighting-example.out
FAILED: chip-efr32-lighting-example.out chip-efr32-lighting-example.out.map
arm-none-eabi-g++ -T../../third_party/connectedhomeip/examples/platform/silabs/efr32/ldscripts/mgm24.ld -march=armv8-m.main+dsp -mcpu=cortex-m33 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Og --specs=nosys.specs --specs=nano.specs -Werror -Wl,--fatal-warnings -fdiagnostics-color -Wl,--gc-sections -Wl,-Map,./chip-efr32-lighting-example.out.map @./chip-efr32-lighting-example.out.rsp -o ./chip-efr32-lighting-example.out
/home/user/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld:../../third_party/connectedhomeip/third_party/silabs/gecko_sdk/protocol/openthread/libs/libsl_openthread_efr32mg2x_gcc.a: file format not recognized; treating as linker script
/home/user/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld:../../third_party/connectedhomeip/third_party/silabs/gecko_sdk/protocol/openthread/libs/libsl_openthread_efr32mg2x_gcc.a:1: syntax error
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

jmartinez-silabs, 2023-05-04:
I'll take a look tomorrow and update you here. I should be able apply a quick fix if need be

jmartinez-silabs, 2023-05-04:
@lboue I also wanted to point out 2 things. I see you are going to the example repo and using the gn gen and ninja commands directly. This is ok. But also leaves you open to some bad configurations if you swap boards and do not clean your out folder as the `args.gni` created in the out for one board might not work for a different one. I believe the last of the issue you hit comes from that

I suggest using the build script. Here is an example

`./scripts/examples/gn_efr32_example.sh examples/lighting-app/silabs/efr32/ out/lighting_app BRD2704A`

closed 2023-05-04 by bzbarsky-apple

lboue, 2023-05-05:
> @lboue I also wanted to point out 2 things. I see you are going to the example repo and using the gn gen and ninja commands directly. This is ok. But also leaves you open to some bad configurations [...]
Thanks for the advice. I tried both methods at first.

PR #26370 (2023-05-04): [Silabs] Fix linkage and some apps build for the Sparkfun board
fixes #26357
fixes #26356
fixes #26350

Two recently added files from Silabs SDK needed to be linked for that board.
This board has no buttons, so I added some `#ifdef` to the button's usage in certain apps.
Removed some include of the buttons header where they were not needed any more thanks to the ongoing silabs platforms abstraction