project-chip/connectedhomeip #26362: [BUG] Java chipDeviceController.onNOCChainGeneration() is forcing use of ICAC when ICAC are optional per the spec.

jonsmirl, 2023-05-04:
### Reproduction steps

https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/CHIPDeviceController-JNI.cpp#L197

Java chipDeviceController.onNOCChainGeneration() is forcing use of ICAC when ICAC are optional per the spec.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

f9339cc

### Platform

android

netscell, 2023-07-14:
noc and rcac cert in onNOCChainGeneration is matter cert or x509 der format?

jonsmirl, 2023-07-14:
Look at the code in the link, it is forcing you to supply an optional parameter.

tnguyen-alarm, 2023-09-08:
I agree that it's strange that it's required. FWIW, I was able to work around it by providing an empty ICAC:

```kotlin
fun getNocChainForDevice(): ControllerParams {
    val selfSignedRcac = ...
    val signedNoc = ...

    return ControllerParams.newBuilder()
        .setRootCertificate(selfSignedRcac)
        .setIntermediateCertificate(byteArrayOf())
        .setOperationalCertificate(signedNoc)
        .setIpk(ipk)
        .build()
}
```