project-chip/connectedhomeip #26363: [BUG] Commissioning hangs without failing

alex-moran-deako, 2023-05-04:
### Reproduction steps

Attempting to do a proof of concept of matter running on our device. I have followed ESP32's example application [here](https://github.com/espressif/esp-matter/tree/main/examples/light) as an example. Matter starts up just fine, and my commissioning device shows up as a "Matter Accessory". After establishing a secured channel, commissioning halts without warning and eventually both devices times out.

Attached are the commissioning logs.

[commission_fail_log.txt](https://github.com/project-chip/connectedhomeip/files/11393027/commission_fail_log.txt)

### Bug prevalence

Repro 100% of the time

### GitHub hash of the SDK that was being used

bd17b9f52

### Platform

esp32

### Platform Version(s)

N/A

### Anything else?

N/A

bzbarsky-apple, 2023-05-04:
@alex-moran-deako Which client are you using?  Have you tried using the chip-tool test application so that you can see logs for both sides of the commissioning?

What I see in the attached log is that the client (whatever it is) is doing a bunch of reads, and those are going fine and then it just stops talking to your device for some reason.  It's hard to tell why it's doing that without having logs from the _client_ side.

alex-moran-deako, 2023-05-04:
I was using an iPad/HomePod as the commissioning device. I agree it's difficult to know what's going on without seeing the client, my thought was since I've seen this client commission other released matter devices then the issue must be on server side.

I will attempt with the chip-tool and report back.

bzbarsky-apple, 2023-05-04:
@alex-moran-deako If you are using iPad/HomePod, you could also follow instructions below to file a feedback assistant ticket.  The sysdiagnose in that will include client logs and whatnot.  If you do that, please let me know what the ticket ID is and I will take a look.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

alex-moran-deako, 2023-05-04:
Hi @bzbarsky-apple 

I've opened a ticket: https://feedbackassistant.apple.com/feedback/12167819

Note when I try to export the analytics off the homepod, the Home app crashes after about a minute timeout. But a video, our logs, and sysdiagnose of the iPad have been uploaded.

alex-moran-deako, 2023-05-08:
I think the issues lies with the idea that the commissioner is not talking to cluster 0x30. In a sample application, I can see the commissioner interacting with 0x30. Whereas with my application with the ported sample code, the commissioner is talking to cluster 0x31. Why is this?

bzbarsky-apple, 2023-05-08:
Cluster 0x30 is General Commissioning.  Every commissioning will interact with that.

Cluster 0x31 is Network Commissioning.  This will only be involved for devices that need network commissioning, which some sample applications may not if they run on-network only.  I can't tell you much more than that without knowing which "sample application" was involved and how it was run.

@alex-moran-deako Thank you for submitting the feedback request ticket.  In the iPad log in that ticket, I see three commissioning attempts that get as far as sending PBKDFParamRequest (one at 14:29, one at 14:42, and one at 14:43).  These happen on exchanges 3967, 62628, and 65361, none of which match the device-side log in the issue (where the PBKDFParamRequest is received on exchange 38463).  Are the client logs from the same instance of the failure as the server log?

That said, the handshake on exchange 3967 is followed by what looks similar to the above-described issue: a bunch of reads, then nothing.  I'm looking into what exactly happened there, but a question for you: Was this on-network commissioning or BLE-based commissioning?  Does your device define any network commissioning clusters, and if so what are their feature maps?

alex-moran-deako, 2023-05-09:
@bzbarsky-apple 
Thanks again for the support.

- "Cluster 0x30 is General Commissioning. Every commissioning will interact with that." 
Is this true for network-only commissioning?

- I believe the device logs and the commissioner logs are not from the exact same instance. I had some questions if I was gathering the Sysdiagnose correctly, so I attempted it several times and I think the log capture was from earlier. If's critical that they are in sync (and I can see why that may be important), I can capture the process again.

- Our application is doing on-network commissioning, and not BLE commissioning. I have disabled CHIPoBLE. This is because ESP's SDK expects to start the BLE stack, which is currently not compatible with my companies codebase. Eventually I'll resolve this, but for the proof of concept I'm hoping to just get away with wifi commissioning. If this is causing issues than it's worth, I can re-prioritize that work but I'm hoping I can kick that can down the road.

- The sample application I am referencing is [this](https://github.com/espressif/esp-matter/tree/main/examples/light), but have slightly altered it be wifi only and set CONFIG_ENABLE_CHIPOBLE=n so that it mirrors my usecase. So I believe this should be commissioning the exact same as my application, only over wifi and it is working.

- "Does your device define any network commissioning clusters, and if so what are their feature maps?"
I am not entirely sure. ESP Matter has wrapped alot of chip code into a nice bundle, but unfortunately I haven't dug too much into what is going on in the code. But given the commissioner is talking to cluster 0x31, I would think it has defined a network commissioning. While I understand this is esp's reference and not chip, for what it's worth this is how I'm utilizing their SDK: 
[vendor_matter.txt](https://github.com/project-chip/connectedhomeip/files/11425990/vendor_matter.txt), which will mirror closely to the [light example.](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L123)

- As a side note, I've looked deeper in parsing the messages between the devices and came across [tracing](https://github.com/project-chip/connectedhomeip/tree/master/examples/common/tracing). Today I worked on consuming this in the two projects and capturing the logs. I still have not gone through the results myself yet here is our failure logs [json_failure.txt](https://github.com/project-chip/connectedhomeip/files/11425921/json_failure.txt) compared to the sample applications success logs: [json_success.txt](https://github.com/project-chip/connectedhomeip/files/11425931/json_success.txt). I'm hoping this gives me (or us) more insight what is being sent between the devices.

bzbarsky-apple, 2023-05-09:
>  Is this true for network-only commissioning?

I'm not sure what you mean by "network-only", but it's true for both on-network (i.e. discovered over existing IP network) and BLE commissioning.

> If's critical that they are in sync (and I can see why that may be important), I can capture the process again.

I think it would be pretty helpful to be able to match up the logs so that it's very clear which part of the (quite long) iOS log is relevant to the exact observed failure.

> Our application is doing on-network commissioning

OK.  That should be fine, but if it's _also_ exposing network commissioning clusters then some versions of iOS may try to figure out how to do network configuration for it even though it does not need that and things will fail.  That might be what's going on here, conceivably.  I assume you are using iOS 16.4?

alex-moran-deako, 2023-05-09:
> "I'm not sure what you mean by "network-only", but it's true for both on-network (i.e. discovered over existing IP network) and BLE commissioning."
Understood. Sorry, still becoming familiar with terminology. 

> "I think it would be pretty helpful to be able to match up the logs so that it's very clear which part of the (quite long) iOS log is relevant to the exact observed failure."
I will capture these logs and reupload them to the ticket on Wednesday Morning.

> " I assume you are using iOS 16.4?"
I'll need to confirm but I believe so. Will do that when I respond with the logs on Wednesday.

It does look like the commissioner is requesting the feature map from network commissioning.
`I (109871) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0031's Attribute 0xFFFC is 0 **********`
0x31 = Networking Commissioning 
0xFFFC = the feature map.
The Network Clusters feature map is defined in 11.8.5. 

Based on the definition, I _think_ bit 0 should be set to 1. The fact it's returning 0 may be putting the commissioner in an undefined state?

I am still not entirely sure why the commissioner is requesting this information and not talking to 0x30. In my light example code, I do not see the commissioner talk to the networking cluster.

Based on Section 5.5 and 5.5.2 of the Matter 1.0 Spec it should be trying to set the fail safe and configure UTC, region, etc. All of this operates on cluster 0x30.

alex-moran-deako, 2023-05-09:
Well this is finally starting to add up a bit, thank you.

It sounds like I'll start digging into espressifs library a bit more. Right now, I am neither setting up or configuring the network commissioning cluster; it's all being done under the hood. I'll see if I can remove the cluster or alter it so that in the feature map it responds with bit 0 set to signal WiFi Networking Interface.

Given we have a good idea where this issue is, would you still like the log capture of synchronized logs between the commissioner and commissionee?

bzbarsky-apple, 2023-05-09:
@alex-moran-deako I wouldn't worry about the log capture for now.  If addressing the network commissioning feature map issue does not resolve things, then we go back to looking at logs.

alex-moran-deako, 2023-05-10:
After running more tests, it seems like I am able to use the chip-tool to reliably commission my device, and then control the matter device.
[chip-tool-commissioning-success.txt](https://github.com/project-chip/connectedhomeip/files/11445356/chip-tool-commissioning-success.txt)
[device-commissioning-success.txt](https://github.com/project-chip/connectedhomeip/files/11445357/device-commissioning-success.txt)

This happens regardless of the value of the feature map of cluster 0x31; it can be 0 or 1 and it still works. I agree  that it being 0 is out of spec of the matter 1.0 spec, but it seems like the chip-tool doesn't care. Maybe this should be checked and errors/warnings be logged that the commissionee is not following spec.

With iOS as the commissioner:
If the feature map of cluster 0x31 is 0, it will not get to step 4 of the commissioning process, arming the failsafe.
If the feature map of cluster 0x31 is 1, it will not get to step 9 of the commissioning process, configuring the operational cert.

This sounds like I should potentially engage apple and gather those synchronized logs @bzbarsky-apple to see what is hanging.

Either way, I'm happy progress has been made after being stuck on this for a few weeks.

bzbarsky-apple, 2023-05-10:
@alex-moran-deako chip-tool has a totally different way of interacting with the network commissioning clusters than iOS, and it does some different things with certificates as well.  At this point if you can gather the synchronized logs and let me know when they are available that would be helpful.

The feature map for network commissioning cluster is handled by the relevant network commissioning driver bits, _if_ that code is reached.  Depending on how espressif implements their stuff, they might be overriding it.

alex-moran-deako, 2023-05-10:
@bzbarsky-apple, I'll grab the synchronized logs while overriding the feature map for the networking cluster. I'll grab this tomorrow morning. Should I create a new feedback ticket since this is failing on a new commissioning step or should I add to the existing on?

alex-moran-deako, 2023-05-10:
> " chip-tool has a totally different way of interacting with the network commissioning clusters than iOS, and it does some different things with certificates as well."

Understood and thanks for that info. Our main motivation for this matter exploration is to add homekit support for our device, so having it work in the Apple ecosystem is the goal for my proof of concept. While the chip-tool successfully commissioning is a hopeful milestone, I'll be continuing this effort until I can commission and control our device via homekit.

bzbarsky-apple, 2023-05-11:
>  Should I create a new feedback ticket since this is failing on a new commissioning step or should I add to the existing on?

@alex-moran-deako Better a new one to make things simpler to track.

alex-moran-deako, 2023-05-11:
Hi @bzbarsky-apple 

I'm no longer seeing consistent results. Commissioning with chip-tool works reliably, nearly with 100% successful. 

With an iPad/HomePod I still see commissioning fail 100% of the time when cluser 0x31 feature map is 0. When I hard code it to 1, commissioning _can_ succeed (roughly 20% of the time). I can definitely see a difference in the device logs between chip-tool/iOS commissioning, although that may be expected. I need to go back to the light example logs and compare.

I want to collect some more data before engaging more support. I am going to close this ticket completely and will re-open a new one on this repo and apples support if appropriate when I have a better understanding of all the moving pieces. 

Thanks again for the assist with figuring out the original issue, the networking commission cluster feature map being invalid.

closed 2023-05-11 by alex-moran-deako

bzbarsky-apple, 2023-05-12:
@alex-moran-deako Sounds good.  If you end up filing a new issue, please just @-mention me in it!