project-chip/connectedhomeip #26387: [BUG] [EFR32] lighting-app for BRD2704A board doesn't work.

lboue, 2023-05-05:
### Reproduction steps

## Issue
Hello,

I build and flash lighting-app to my BRD2704A board and it doesn't work. 
I am getting `App Critical Error: 172 ` from the logs:

````
JLinkRTTClient
SEGGER J-Link V7.88 - Real time terminal output
Silicon Labs J-Link OB compiled Sep 30 2022 13:05:09 V1.0, SN=449050146
Process: JLinkExe
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]SiLabs-Light starting
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Init CHIP Stack
[00:00:00.000][silabs ]!!!!!!!!!!!! App Critical Error: 172 !!!!!!!!!!!
```

## Reproduce steps

### flash lighting-app for BRD2704A
```
~/connectedhomeip$ ./scripts/examples/gn_efr32_example.sh ./examples/lighting-app/silabs/efr32/ ./out/lighting-app  BRD2704A
[970/976] touch obj/lighting_app.hex.stamp
[971/976] touch obj/lighting_app.flash_executable.image.stamp
[972/976] touch obj/lighting_app.flash_executable.flashing.stamp
[973/976] touch obj/lighting_app.flash_executable.stamp
[974/976] touch obj/lighting_app.stamp
[975/976] touch obj/efr32.stamp
[976/976] touch obj/default.stamp
+ arm-none-eabi-size -A ./out/lighting-app/BRD2704A/chip-efr32-lighting-example.out
./out/lighting-app/BRD2704A/chip-efr32-lighting-example.out  :
section               size        addr
.text               918000   134242304
.ARM.exidx               8   135160304
.copy.table             12   135160312
.zero.table              0   135160324
.data                 2588   536870912
.bss                139604   536873504
.nvm                 40960   135160324
.ARM.attributes         54           0
.comment               158           0
.debug_line_str        732           0
.debug_info       42547782           0
.debug_abbrev      1561293           0
.debug_loclists    2026778           0
.debug_aranges      133984           0
.debug_rnglists     127489           0
.debug_line        3987774           0
.debug_str         3556515           0
.debug_frame        435400           0
Total             55479131
```

### flash bootloader
```
~/connectedhomeip$ ~/commander/commander flash ~/SiliconLabs/bootloader_binaries/bootloader-storage-internal-single-512k-BRD2704A-gsdk4.2.s37
Parsing file /home/lboue/SiliconLabs/bootloader_binaries/bootloader-storage-internal-single-512k-BRD2704A-gsdk4.2.s37...
Writing 16384 bytes starting at address 0x08000000
Comparing range 0x08000000 - 0x08003FFF (16 KB)
Flashing completed successfully!
DONE
```

### Getting logs from the board

```
JLinkRTTClient
SEGGER J-Link V7.88 - Real time terminal output
Silicon Labs J-Link OB compiled Sep 30 2022 13:05:09 V1.0, SN=449050146
Process: JLinkExe
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]SiLabs-Light starting
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Init CHIP Stack
[00:00:00.000][silabs ]!!!!!!!!!!!! App Critical Error: 172 !!!!!!!!!!!
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

3d2170808186a16ac31d8f0dcf0fa1aeeedae071

### Platform

efr32

bzbarsky-apple, 2023-05-05:
@jmartinez-silabs

jmartinez-silabs, 2023-05-05:
My first guess would be a corrupt nvm3 section. It could happen with flashing other silabs app (non-matter) to that same board for example.

Try to use commander to erase the chip
[image]

You will have to reflash a bootloader also on your chip. you can fin the binary for the bootloader of your board here
`third_party/silabs/matter_support/matter/efr32/bootloader_binaries/bootloader-storage-internal-single-512k-BRD2704A-gsdk4.2.s37`

the reflash your light app

Let me know if that helps.

lboue, 2023-05-05:
It works after Erasing. Thanks for your help @jmartinez-silabs.