project-chip/connectedhomeip #26388: [BUG] [EFR32] Fail to build lighting-app with pigweed RPC

lboue, 2023-05-05:
### Reproduction steps

Fail to build lighting-app with pigweed RPC

This is an issue here:
examples/lighting-app/silabs/efr32/third_party/connectedhomeip/examples/platform/silabs/Rpc.cpp:85:31
`error: 'class AppTask' has no member named 'ButtonEventHandler'; did you mean 'static void `

```
../../../examples/lighting-app/silabs/efr32/third_party/connectedhomeip/examples/platform/silabs/Rpc.cpp: In member function 'virtual pw::Status chip::rpc::Efr32Button::Event(const chip_rpc_ButtonEvent&, pw_protobuf_Empty&)':
../../../examples/lighting-app/silabs/efr32/third_party/connectedhomeip/examples/platform/silabs/Rpc.cpp:85:31: error: 'class AppTask' has no member named 'ButtonEventHandler'; did you mean 'static void BaseApplication::FunctionEventHandler(AppEvent*)'? (not accessible from this context)
   85 |         AppTask::GetAppTask().ButtonEventHandler(request.idx /* PB 0 or PB 1 */, request.pushed);
      |                               ^~~~~~~~~~~~~~~~~~
In file included from ../../../examples/lighting-app/silabs/efr32/include/AppTask.h:30,
                 from ../../../examples/lighting-app/silabs/efr32/third_party/connectedhomeip/examples/platform/silabs/Rpc.cpp:19:
../../../examples/lighting-app/silabs/efr32/third_party/connectedhomeip/examples/platform/silabs/efr32/BaseApplication.h:155:17: note: declared protected here
  155 |     static void FunctionEventHandler(AppEvent * aEvent);
      |                 ^~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
ninja: build stopped: subcommand failed.
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

3d2170808186a16ac31d8f0dcf0fa1aeeedae071

### Platform

efr32

lboue, 2023-05-05:
> Ah, I missed that one. There are no buttons on your Sparkfun board(BRD2704) therefore no handler. Pigweed rpc won't work on that board currently.
> 
> I'll look if it is possible to build the pw-rpc support without the button.

And if I connect buttons to the GPIO ports?

jmartinez-silabs, 2023-05-12:
@lboue didn't have time to check yet.

Somyounhave a usecase for PW rpc with spake fun?

lboue, 2024-03-18:
It works now:

```
user@ubuntu:~/connectedhomeip$ ./scripts/examples/gn_silabs_example.sh examples/lighting-app/silabs/ out/lighting_app_rpc BRD2704A 'import("//with_pw_rpc.gni")'
user@ubuntu:~/connectedhomeip$ ls ./out/lighting_app_rpc/BRD2704A/
args.gn        build.ninja.stamp      gen                           matter-silabs-lighting-example.flash.py  matter-silabs-lighting-example.out.map  protocol_buffer  relative_path_transformations.json  zap_gen.lock
build.ninja    compile_commands.json  lib                           matter-silabs-lighting-example.hex       matter-silabs-lighting-example.s37      python           silabs_firmware_utils.py
build.ninja.d  firmware_utils.py      lighting_app.flashbundle.txt  matter-silabs-lighting-example.out       obj                                     python-venv      toolchain.ninja
```

closed 2024-08-28 by lboue