project-chip/connectedhomeip #26407: [BUG] iOS 16.4.1 tries to read" bridged basic information cluster" on root endpoint on a bridged node

Apollon77, 2023-05-07:
### Reproduction steps

Construct a bridge as shown in the graphic in Core Specs 1.0 pg 476 and pair it with iOS Home.

It tries to read bridged basic information cluster data (mainly reachable attribute) on the root cluster too and continues doing that.

Here a log from matter.js:
```
2023-04-30 14:39:55.166 DEBUG MessageExchange new MessageExchange 1 31982 300 5000 5
2023-04-30 14:39:55.166 DEBUG MessageExchange onMessageReceived 1 id:44990/31982/99410684 t:1/2 reqAck:true payload: 153600172402002403392404111818290324ff0118
2023-04-30 14:39:55.167 DEBUG InteractionProtocol Received read request from udp://fe80::185d:6f90:4c5d:85de%en0:51981 on session secure/44990: MA-rootdevice(0x0)/unknown(0x39)/0x11, isFabricFiltered=true
2023-04-30 14:39:55.167 DEBUG InteractionProtocol Read from udp://fe80::185d:6f90:4c5d:85de%en0:51981 on session secure/44990: MA-rootdevice(0x0)/unknown(0x39)/0x11 unsupported path
2023-04-30 14:39:55.167 DEBUG ExchangeManager sending id:44990/31982/1627674264 t:1/5 acked:99410684 reqAck:true payload: 153601153500370024020024033924041118350124008618181818280424ff0118
2023-04-30 14:39:55.184 DEBUG MessageExchange onMessageReceived 1 id:44990/31982/99410685 t:1/1 acked:1627674264 reqAck:true payload: 1524000024ff0118
2023-04-30 14:39:55.184 DEBUG ExchangeManager sending id:44990/31982/1627674265 t:0/16 acked:99410685 reqAck:false payload: 
```

Is this an iOS bug or a documentation isssue and we need to add the cluster also to root endpoint?

@bzbarsky-apple maybe something you know ....

### Bug prevalence

anytime

### GitHub hash of the SDK that was being used

n/a

### Platform

other

bzbarsky-apple, 2023-05-08:
@Apollon77 When you do file that feedback assistant ticket, please mention the ticket ID here and I will follow up on it.

Apollon77, 2023-05-08:
Ok. Thank you for that clarification. ;-) I will see that I or @vves file a ticket.

woody-apple, 2023-06-23:
Closing this, as this should be followed up through feedback assistant.

closed 2023-06-23 by woody-apple