project-chip/connectedhomeip #26432: Make script to generate a basic RevocationSet from TestNet

tcarmelveilleux, 2023-05-08:
The algorithm specified in section "6.2.4.1. Conceptual algorithm for revocation set construction" should be built and usable against TestNet schema, MainNet schema, and against a file containing an array of JSON records equivalent to getting all revocation distribution points from TestNet or MainNet.

The output file should have a JSON format of:

```
[
  { 
    "type": "revocation_set",
    "issuer_subject_key_id": "<issuer subject key ID as uppercase hex, 20 bytes>",
    "issuer_name": "<ASN.1 SEQUENCE of Issuer of the CRL as base64>",
    "revoked_serial_numbers:
    [
      "serial1 bytes as base64",
      "serial2 bytes as base64"
    ]
  }
]
```

This should be built as a Python script, using dependencies in the repos. This will be usable to provide an input to device attestation revocation "basic proof of concept" implementation.

See https://github.com/project-chip/connectedhomeip/blob/master/credentials/fetch-paa-certs-from-dcl.py for an example of a script making use of DCL to download DCL schema entries.

tcarmelveilleux, 2023-11-27:
Assigning to Vijay, to determine effort. Please re-assign to me if this is too large.

shubhamdp, 2025-01-23:
script is present here: https://github.com/project-chip/connectedhomeip/blob/master/credentials/generate_revocation_set.py.

This can be marked as Fixed!

closed 2025-01-24 by tcarmelveilleux