project-chip/connectedhomeip #26438: [BUG] [bridge-app] no matter what attribute, write failed, got an UNSUPPORTED_WRITE respond.

weihao042122, 2023-05-09:
### Reproduction steps

bridge-app/linux ; used the chip-tool to write the onoff cluster's attribute, but got an "UNSUPPORTED_WRITE" respond.

./chip-tool onoff write on-time  11 254 3
<details>
<summary>here is the chip-tool logs:</summary>
<pre><code>[1683598115.961612][9265:9265] CHIP:TOO: Command: onoff write on-time 11 254 3
[1683598115.962221][9265:9268] CHIP:TOO: Sending command to node 0xfe
[1683598115.962517][9265:9268] CHIP:CSM: FindOrEstablishSession: PeerId = [1:00000000000000FE]
[1683598115.962536][9265:9268] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1683598115.962554][9265:9268] CHIP:DIS: Found an existing secure session to [1:00000000000000FE]!
[1683598115.962562][9265:9268] CHIP:DIS: OperationalSessionSetup[1:00000000000000FE]: State change 1 --> 5
[1683598115.962585][9265:9268] CHIP:TOO:        cluster 0x0000_0006, attribute: 0x0000_4001, endpoint 3
[1683598115.962609][9265:9268] CHIP:DMG: WriteClient moving to [AddAttribu]
[1683598115.962663][9265:9268] CHIP:EM: <<< [E:21677i S:15776 M:204916501] (S) Msg TX to 1:00000000000000FE [DFFA] --- Type 0001:06 (IM:WriteRequest)
[1683598115.962674][9265:9268] CHIP:IN: (S) Sending msg 204916501 on secure session with LSID: 15776
[1683598115.962766][9265:9268] CHIP:DMG: WriteClient moving to [AwaitingRe]
[1683598116.173688][9265:9268] CHIP:EM: >>> [E:21677i S:15776 M:158159013 (Ack:204916501)] (S) Msg RX from 1:00000000000000FE [DFFA] --- Type 0001:07 (IM:WriteResponse)
[1683598116.173789][9265:9268] CHIP:EM: Found matching exchange: 21677i, Delegate: 0x7fe2f4022580
[1683598116.173806][9265:9268] CHIP:EM: Rxd Ack; Removing MessageCounter:204916501 from Retrans Table on exchange 21677i
[1683598116.173817][9265:9268] CHIP:DMG: WriteClient moving to [ResponseRe]
[1683598116.173831][9265:9268] CHIP:DMG: WriteResponseMessage =
[1683598116.173835][9265:9268] CHIP:DMG: {
[1683598116.173838][9265:9268] CHIP:DMG:        AttributeStatusIBs =
[1683598116.173844][9265:9268] CHIP:DMG:        [
[1683598116.173847][9265:9268] CHIP:DMG:                AttributeStatusIB =
[1683598116.173851][9265:9268] CHIP:DMG:                {
[1683598116.173854][9265:9268] CHIP:DMG:                        AttributePathIB =
[1683598116.173859][9265:9268] CHIP:DMG:                        {
[1683598116.173862][9265:9268] CHIP:DMG:                                Endpoint = 0x3,
[1683598116.173866][9265:9268] CHIP:DMG:                                Cluster = 0x6,
[1683598116.173870][9265:9268] CHIP:DMG:                                Attribute = 0x0000_4001,
[1683598116.173874][9265:9268] CHIP:DMG:                        }
[1683598116.173879][9265:9268] CHIP:DMG:
[1683598116.173882][9265:9268] CHIP:DMG:                        StatusIB =
[1683598116.173886][9265:9268] CHIP:DMG:                        {
[1683598116.173890][9265:9268] CHIP:DMG:                                status = 0x88 (UNSUPPORTED_WRITE),
[1683598116.173893][9265:9268] CHIP:DMG:                        },
[1683598116.173897][9265:9268] CHIP:DMG:
[1683598116.173900][9265:9268] CHIP:DMG:                },
[1683598116.173905][9265:9268] CHIP:DMG:
[1683598116.173911][9265:9268] CHIP:DMG:        ],
[1683598116.173916][9265:9268] CHIP:DMG:
[1683598116.173919][9265:9268] CHIP:DMG:        InteractionModelRevision = 1
[1683598116.173922][9265:9268] CHIP:DMG: }
[1683598116.173943][9265:9268] CHIP:DMG: WriteClient moving to [AwaitingDe]
[1683598116.173950][9265:9268] CHIP:TOO: Response Failure: IM Error 0x00000588: General error: 0x88 (UNSUPPORTED_WRITE)
[1683598116.173959][9265:9268] CHIP:TOO: Run command failure: IM Error 0x00000588: General error: 0x88 (UNSUPPORTED_WRITE)
[1683598116.173988][9265:9268] CHIP:EM: <<< [E:21677i S:15776 M:204916502 (Ack:158159013)] (S) Msg TX to 1:00000000000000FE [DFFA] ---
Type 0000:10 (SecureChannel:StandaloneAck)
[1683598116.174354][9265:9268] CHIP:IN: (S) Sending msg 204916502 on secure session with LSID: 15776
[1683598116.174423][9265:9268] CHIP:EM: Flushed pending ack for MessageCounter:158159013 on exchange 21677i
</code></pre>
</details>

<details>
<summary>here is the bridge-app logs:</summary>
<pre><code>CHIP:EM: >>> [E:21682r S:22278 M:136356386] (S) Msg RX from 3:000000000001B669 [DFFA] --- Type 0001:06 (IM:WriteRequest)
CHIP:EM: Handling via exchange: 21682r, Delegate: 0x402add1c
CHIP:IM: Received Write request
CHIP:DMG: IM WH moving to [Initialized]
CHIP:DMG: WriteRequestMessage =
CHIP:DMG: {
CHIP:DMG:       suppressResponse = false,
CHIP:DMG:       timedRequest = false,
CHIP:DMG:       AttributeDataIBs =
CHIP:DMG:       [
CHIP:DMG:               AttributeDataIB =
CHIP:DMG:               {
CHIP:DMG:                       AttributePathIB =
CHIP:DMG:                       {
CHIP:DMG:                               Endpoint = 0x3,
CHIP:DMG:                               Cluster = 0x6,
CHIP:DMG:                               Attribute = 0x0000_4001,
CHIP:DMG:                       }
CHIP:DMG:
CHIP:DMG:                       Data = 11,
CHIP:DMG:               },
CHIP:DMG:
CHIP:DMG:       ],
CHIP:DMG:
CHIP:DMG:       moreChunkedMessages = false,
CHIP:DMG:       InteractionModelRevision = 1
CHIP:DMG: },
CHIP:DMG: IM WH moving to [AddStatus]
CHIP:EM: <<< [E:21682r S:22278 M:141484854 (Ack:136356386)] (S) Msg TX to 3:000000000001B669 [DFFA] --- Type 0001:07 (IM:WriteResponse)CHIP:IN: (S) Sending msg 141484854 on secure session with LSID: 22278
CHIP:DMG: IM WH moving to [Sending]
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:EM: >>> [E:21682r S:22278 M:136356387 (Ack:141484854)] (S) Msg RX from 3:000000000001B669 [DFFA] --- Type 0000:10 (SecureChannel:StandaloneAck)
CHIP:EM: Found matching exchange: 21682r, Delegate: (nil)
CHIP:EM: Rxd Ack; Removing MessageCounter:141484854 from Retrans Table on exchange 21682r</code></pre>
</details>

off-wait-time, start-up-on-off, also get the "UNSUPPORTED_WRITE" respond

### Bug prevalence

Repro 100% of the time

### GitHub hash of the SDK that was being used

f994daebd5dd7645268d5a1785ede1c0bcb2b3cb

### Platform

raspi

bzbarsky-apple, 2023-05-09:
@weihao042122 Which exact bridge-app is this?  How are the relevant endpoints/clusters configured in that app?  Are the attributes marked as writable?

weihao042122, 2023-05-09:
@bzbarsky-apple 
it is the "CHIP Linux Bridge Example", and using the "Dynamic Endpoints".
here is the declear of the cluster/attribute config:
```
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(onOffAttrs)
	DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::OnOff::Id,BOOLEAN,1,0),
	DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::GlobalSceneControl::Id,BOOLEAN, 1,0),
	DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::OnTime::Id, INT16U, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::OffWaitTime::Id, INT16U, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::FeatureMap::Id,BITMAP32,4, 0),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();
```
And How to make the attributes as writable?

weihao042122, 2023-05-09:
problem solved . make the attributes  as writeable.

thanks

closed 2023-05-09 by weihao042122

nmtoan91, 2023-05-15:
@bzbarsky-apple @weihao042122 
can you teach me how to set an attribute writeable?  
I'm making a bridged app for thermostat, I can read the system-mode but can't write

nmtoan91, 2023-05-15:
this is a part of my zap file but I got UNSUPPORTED_WRITE for writing system-mode

```
readonly attribute nullable int16s localTemperature = 0;
  readonly attribute int16s absMinHeatSetpointLimit = 3;
  readonly attribute int16s absMaxHeatSetpointLimit = 4;
  readonly attribute int16s absMinCoolSetpointLimit = 5;
  readonly attribute int16s absMaxCoolSetpointLimit = 6;
  attribute int16s occupiedCoolingSetpoint = 17;
  attribute int16s occupiedHeatingSetpoint = 18;
  attribute access(write: manage) int16s minHeatSetpointLimit = 21;
  attribute access(write: manage) int16s maxHeatSetpointLimit = 22;
  attribute access(write: manage) int16s minCoolSetpointLimit = 23;
  attribute access(write: manage) int16s maxCoolSetpointLimit = 24;
  attribute access(write: manage) int8s minSetpointDeadBand = 25;
  attribute access(write: manage) ThermostatControlSequence controlSequenceOfOperation = 27;
  attribute access(write: manage) enum8 systemMode = 28;
  readonly attribute command_id generatedCommandList[] = 65528;
  readonly attribute command_id acceptedCommandList[] = 65529;
  readonly attribute event_id eventList[] = 65530;
  readonly attribute attrib_id attributeList[] = 65531;
  readonly attribute bitmap32 featureMap = 65532;
  readonly attribute int16u clusterRevision = 65533;
```

bzbarsky-apple, 2023-05-15:
@nmtoan91 What does your actual endpoint configuration looks like in your code, for the endpoint that has the SystemMode attribute?

nmtoan91, 2023-05-15:
@bzbarsky-apple 
I declare the endpoint by:

```
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(thermostatAttrs_)
     DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::LocalTemperature::Id, INT16S, 2, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::OutdoorTemperature::Id, INT16S, 2, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::Occupancy::Id, BITMAP8, 1, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::ControlSequenceOfOperation::Id, ENUM8, 1, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::SystemMode::Id, ENUM8, 1, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::FeatureMap::Id, BITMAP32, 4, 0),     /* FeatureMap */
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

constexpr CommandId thermostatIncomingCommands_[] = {
    kInvalidCommandId,
};

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(fancontrolAttrs_)
     DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::FanMode::Id, ENUM8, 1, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::FanModeSequence::Id, ENUM8, 1, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::PercentSetting::Id, INT8U, 1, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::PercentCurrent::Id, INT8U, 1, 0), 
     DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::FeatureMap::Id, BITMAP32, 4, 0),     /* FeatureMap */
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(heatingColingClusters_)
    DECLARE_DYNAMIC_CLUSTER(OnOff::Id, onOffAttrs_, onOffIncomingCommands_, nullptr),
    DECLARE_DYNAMIC_CLUSTER(Thermostat::Id, thermostatAttrs_, thermostatIncomingCommands_, nullptr),
    DECLARE_DYNAMIC_CLUSTER(FanControl::Id, fancontrolAttrs_, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(LevelControl::Id, dimmableAttrs_, levelControlIncomingCommands_, nullptr),
    DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs_, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasicInformation::Id, bridgedDeviceBasicAttrs_, nullptr,nullptr) 
DECLARE_DYNAMIC_CLUSTER_LIST_END;
DECLARE_DYNAMIC_ENDPOINT(heatingColingEndpoint_, heatingColingClusters_); 
DataVersion heatingColingDataVersions_[ArraySize(heatingColingClusters_)]; 
```

weihao042122, 2023-05-15:
@nmtoan91 declare the dynamic cluster like this:
```
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OnLevel::Id, INT8U, 1, ZAP_ATTRIBUTE_MASK(WRITABLE) | ZAP_ATTRIBUTE_MASK(NULLABLE)),
```

nmtoan91, 2023-05-15:
@weihao042122 Thanks, you saved my day :D

bzbarsky-apple, 2023-05-16:
Right, `ZAP_ATTRIBUTE_MASK(WRITABLE)` for the ones that should be writable.  `ZAP_ATTRIBUTE_MASK(NULLABLE)` only if they are nullable, which is independent of whether they are writable.