project-chip/connectedhomeip #26440: [ESP32] Default value of attributes of window covering with Apple.

jadhavrohit924, 2023-05-09:
### Reproduction steps

In spec default value of `CurrentPositionLiftPercentage, TargetPositionLiftPercent100ths, CurrentPositionLiftPercent100ths` is `null`. When we set these values as `null` and use with apple after commissioning `device is not reachable` but when we set these attributes values to `0` as default value, `device is rechable` and works fine. 
Also in the `examples/window-app` the default value is `0`. 

What should be the default values of these attributes?

### Bug prevalence

Always 

### GitHub hash of the SDK that was being used

None

### Platform

esp32

dhrishi, 2023-05-09:
@bzbarsky-apple Can you please take a look at this issue?

bzbarsky-apple, 2023-05-09:
"default" in the spec means "the value that should be assumed if a specification algorithm needs the value and the value is not provided".  It does _not_ mean "the value a device should be configured to send by default".  It definitely does not mean "what you set the default column in ZAP to".  This is a common confusion, unfortunately.

With that said, per spec for window covering's Current* values:

> The null value indicates that the current position is unknown, e.g. calibration is needed

and for the Target* values:

> The null value indicates that the value is unavailable, e.g. no target position has been set.

So null for a target position is perfectly reasonable, if there is no target position set.  null for a current position is reasonable if the current position is not in fact known.  I don't know whether that's the situation you are in or not.

In either case, "device is not reachable" seems like an odd way on Home's part to react to "position is not known.  Can you please file a feedback ticket per the instructions below, with the relevant logs, and then mention the ticket ID here so I can follow up on it?

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

woody-apple, 2023-06-23:
Closing this, as this should be followed up with Apple directly via feedback assistant.

closed 2023-06-23 by woody-apple