project-chip/connectedhomeip #26445: [TC-MOD-1.1]ModeSelect cluster -- ClusterRevision value missmatch

Survensa, 2023-05-09:
**Problem**
As per Spec and test plan, cluster revision value should be 2 but currently getting cluster revision value as 1 in SDK .

**Expected behavior**: ClusterRevision value as 2

**Actual behavior** : ClusterRevision value as 1

**Steps to reproduce**
./chip-tool modeselect read cluster-revision 1 1

**System configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : Onnetwork
Commit-ID:6c0c5ea6d2b8f38c6b36b83195b6ee55384bb599
Dut application: all-clusters-app

**Test-Plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/modeselect.adoc

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/ModeSelect.adoc

**Log**
[TC-MOD-1.1.txt](https://github.com/project-chip/connectedhomeip/files/11429930/TC-MOD-1.1.txt)

bzbarsky-apple, 2023-05-09:
The SDK does not yet implement revision 2 of the cluster. This is not just about bumping the revision number; support is also needed for ChangeToModeWithStatus, the EXT feature, etc.

rbultman, 2023-07-19:
The mode select cluster has almost completely reverted back to its original state prior to the additions being moved into the modebase cluster.  The revision is still 2, however, but the only substantive change made was to remove the nullable quality from MfgCode.  Not sure how that affects your comment @bzbarsky-apple.

bzbarsky-apple, 2023-07-20:
SDK implements revision 2, since it had MfgCode non-nullable to start with.  We should bump the revision in the SDK...

andy31415, 2023-08-29:
Need just a version bump, assigning to myself (expecting just a zap file change)

bzbarsky-apple, 2023-08-31:
> (expecting just a zap file change)

The XML also needs to be changed.

closed 2023-09-11 by mergify[bot]