project-chip/connectedhomeip #26451: [BUG]

aravindone, 2023-05-09:
### Reproduction steps

Dear devs .

Im facing a issue on the commissioning part of esp 32 with client has chip-tool,
While i pass 
$ ./chip-tool pairing ble-wifi <node_id> <ssid> <password> <pin_code> <discriminator>

im facing opcreds based  error.

Here are the log files 

[https://github.com/aravindone/codes/blob/main/Qr%20code%20feed%20log%20uncommission%20state](url)
[https://github.com/aravindone/codes/blob/main/Qr%20code%20feed%20log%20commission%20state](url)

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

ac810b186ed4f954323afd804cd4d1465ed8b6f2

### Platform

esp32

### Platform Version(s)

1.1.0-1.1

bzbarsky-apple, 2023-05-09:
@aravindone Those log file links both link to https://github.com/project-chip/connectedhomeip/issues/url which is not very helpful.

aravindone, 2023-05-09:
> @aravindone Those log file links both link to https://github.com/project-chip/connectedhomeip/issues/url which is not very helpful.

now i have uploaded the logs @bzbarsky-apple

bzbarsky-apple, 2023-05-09:
@aravindone Link was still broken, but I fixed it....

What that log is showing is that the BLE scan starts, no Matter devices are discovered, and commissioning times out.:
```
[1683639733.670439][33684:33687] CHIP:CTL: Commissioning discovery over BLE failed: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/BLEManagerImpl.cpp:787: CHIP Error 0x00000032: Timeout
```
There's nothing there about opcreds.

aravindone, 2023-05-10:
> @aravindone Link was still broken, but I fixed it....
>
> What that log is showing is that the BLE scan starts, no Matter devices are discovered, and commissioning times out.:
>
> ``` [...]

@bzbarsky-apple i have missed this log now you can have look

bzbarsky-apple, 2023-05-11:
OK, so in the new log what we have is commissioning going along until chip-tool sends a AddOrUpdateWiFiNetwork command and the server responds with INVALID_COMMAND.  Then commissining fails.

On the server side, all the log says for this part is:
```
I (15888) chip[EM]: >>> [E:45714r M:106803610] (S) Msg RX from 1:FFFFFFFB00000000 [B6E0] --- Type 0001:08 (IM:InvokeCommandRequest)
I (15898) chip[EM]: <<< [E:45714r M:154830294] (S) Msg TX to 1:FFFFFFFB00000000 [B6E0] --- Type 0001:09 (IM:InvokeCommandResponse)
```

Why the device decided to respond this way... I don't know.  The esp32 all-clusters-app should have a working network commissioning implementation, last I checked, which this device certainly does not seem to.

aravindone, 2023-05-15:
@bzbarsky-apple which tool is giving this issue ? . is the problem is with the chip-tool or from the Esp 32 all cluster app ?

bzbarsky-apple, 2023-05-16:
The problem is from the device side, so from the all-clusters-app if that's what the device is running.

aravindone, 2023-05-16:
Hi @bzbarsky-apple  This time i tried with ligthing-app example im getting failed with 

`E (240190) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (240190) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!`

here are the logs.

https://raw.githubusercontent.com/aravindone/aravindone/main/chiptool_logs

https://raw.githubusercontent.com/aravindone/aravindone/main/comissioning%20failure%20(ble-wifi)%20log%20esp

bzbarsky-apple, 2023-05-16:
@aravindone That log shows that chip-tool sent ConnectNetwork and the device responded with NetworkNotFound. Then chip-tool gave up on the commissioning, and eventually the fail-safe timed out.

What's odd is that the device log sure seems to show it joining a network, so why did it respond with NetworkNotFound?

aravindone, 2023-05-28:
im closing the ticket because the issue was resolved using direct ubuntu laptop

closed 2023-05-28 by aravindone