project-chip/connectedhomeip #26453: Matter-over-Thread unresponsive

djandrew2005, 2023-05-09:
I am using Google Nest Hub and I have connected a matter-over-thread device. The device works with Google Home just fine but when I try to pair with other platforms the device becomes unresponsive on the other platforms but still working in Google Home.

I have tried adding it to Home Assistant from an Android device (Thread Credentials did transfer to Home Assistant while adding) and to Apple Home with iOS 16.5 (they added support in this version after my request for third party Thread border routers)

bzbarsky-apple, 2023-05-09:
@djandrew2005 If you are running into this problem with Apple Home, can you please follow instructions below to file a feedback assistant ticket that will have relevant logs?   Mention the ticket id here and I will follow up on it.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

djandrew2005, 2023-05-09:
FB12179363

djandrew2005, 2023-05-10:
@bzbarsky-apple This is also a bug you need to check out 
FB12182031

bzbarsky-apple, 2023-05-11:
> FB12179363

For that one, I am not seeing any resident (AppleTV/Homepod) logs.  Are you using the iPad as the resident there?

> FB12182031

I've forwarded that to the Home folks.

bzbarsky-apple, 2023-05-11:
> I've forwarded that to the Home folks.

That said, @djandrew2005 I don't know what "doesn’t always save the device to keychain for easily add to other platform. It also causes errors when turning on pairing mode" means....

djandrew2005, 2023-05-14:
Thanks. I added new logs on the other report. You can see if it helps

bzbarsky-apple, 2023-05-26:
@djandrew2005 Sorry for the lag here.  Which one is the "other report"?  Which log files, if you recall, did you add?

woody-apple, 2023-06-23:
Closing this as this feedback should be tracked through the feedback tracker

closed 2023-06-23 by woody-apple