project-chip/connectedhomeip #26466: [BUG] [chip-tool] cannot build for armhf (32bit ARM)

mmt-10, 2023-05-10:
### Reproduction steps

As a prerequisite, I added `ports.ubuntu.com` to `/etc/apt/sources.list.d/ports_armhf.list`.

```bash
$ sudo dpkg --add-architecture armhf
$ sudo apt update
$ sudo apt install crossbuild-essential-armhf
$ sudo apt install libssl-dev:armhf
$ sudo apt install libglib2.0-dev:armhf
$ cd ${MATTER_SDK}
$ source scripts/activate.sh
$ cd examples/chip-tool
$ gn gen out/debug --args='target_os="linux" target_cpu="arm"'
$ ninja -C out/debug
```

**LOG**
[chip-tool-armhf-build.log](https://github.com/project-chip/connectedhomeip/files/11437723/chip-tool-armhf-build.log)

> /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h

I set `target_cpu="arm"`, but ninja accessed `x86_64-linux-gnu` directory.

and it maybe root cause is 64bit variable.

Can I resolve it?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 4088a77f557e8571a39338fad51a1d8eb0131d79 (HEAD, tag: v1.0.0.2)

### Platform

other

mmt-10, 2023-05-10:
resolve it

```bash
$ export PKG_CONFIG_PATH="/usr/lib/arm-linux-gnueabihf/pkgconfig"
```

bzbarsky-apple, 2023-05-10:
Right, the glib path comes from pkg-config, and that will do what you tell it to do....

closed 2023-05-10 by bzbarsky-apple

xlfdan, 2023-07-14:
@mmt-10 , can you show the full content of ports_armhf.list and sources.list files? Thank you very much.

xlfdan, 2023-07-14:
ubuntu@ubuntu-VM:~$ sudo apt install libssl-dev:armhf

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

Some packages could not be installed. This may mean that you have

requested an impossible situation or if you are using the unstable

distribution that some required packages have not yet been created

or been moved out of Incoming.

The following information may help to resolve the situation:

The following packages have unmet dependencies:

 libssl1.1:armhf : Depends: libc6:armhf (>= 2.28) but it is not going to be installed

E: Unable to correct problems, you have held broken packages.

ubuntu@ubuntu-VM:~$ 

ubuntu@ubuntu-VM:~$ sudo apt install libssl1.1:armhf

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

Some packages could not be installed. This may mean that you have

requested an impossible situation or if you are using the unstable

distribution that some required packages have not yet been created

or been moved out of Incoming.

The following information may help to resolve the situation:

The following packages have unmet dependencies:

 libssl1.1:armhf : Depends: libc6:armhf (>= 2.28) but it is not going to be installed

E: Unable to correct problems, you have held broken packages.

ubuntu@ubuntu-VM:~$ 

ubuntu@ubuntu-VM:~$ sudo apt install libc6:armhf

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

Some packages could not be installed. This may mean that you have

requested an impossible situation or if you are using the unstable

distribution that some required packages have not yet been created

or been moved out of Incoming.

The following information may help to resolve the situation:

[... 160 lines cut ...]
              PreDepends: libc6 (>= 2.34) but it is not installable

              PreDepends: libcap-ng0 (>= 0.7.9) but it is not going to be installed

              PreDepends: libmount1 (>= 2.37.2) but it is not going to be installed

              PreDepends: libpam0g (>= 0.99.7.1) but it is not going to be installed

              PreDepends: libselinux1 (>= 3.1~) but it is not going to be installed

              PreDepends: libsmartcols1 (>= 2.34) but it is not going to be installed

              PreDepends: libsystemd0 but it is not going to be installed

              PreDepends: libtinfo6 (>= 6) but it is not going to be installed

              PreDepends: libudev1 (>= 183) but it is not going to be installed

              PreDepends: libuuid1 (>= 2.16) but it is not going to be installed

              PreDepends: zlib1g (>= 1:1.1.4) but it is not going to be installed

E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

ubuntu@ubuntu-VM:~$

mmt-10, 2023-07-17:
I'm sorry, I had vacation.

Please refer to below quote.
`ports_armhf.list` is here.

However, I gave up on cross-build.
because, I cannot resolve the library(glibc3.0 and etc) dependencies for the target board.
If you can resolve the library dependencies, please let me know.

> deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy main restricted universe multiverse
> deb-src [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy main restricted universe multiverse
> 
> deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy-updates main restricted universe multiverse
> deb-src [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy-updates main restricted universe multiverse
> 
> deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy-backports main restricted universe multiverse
> deb-src [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy-backports main restricted universe multiverse
> 
> deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy-security main restricted universe multiverse
> deb-src [arch=armhf] http://ports.ubuntu.com/ubuntu-ports jammy-security main restricted universe multiverse
>