project-chip/connectedhomeip #26468: [BUG] need to commission matter device through IP

aravindone, 2023-05-10:
### Reproduction steps

Step 1 : all cluster app is installed in esp32 devkit c and wifi ssid and password hardcoded in esp-Idf framework
Step 2 :  try to connect with python-repl tool using ip address

[image]

https://raw.githubusercontent.com/aravindone/aravindone/main/log%20for%20python%20repl%20with%20wifi%20credentials

### Bug prevalence

regular

### GitHub hash of the SDK that was being used

 ac810b1

### Platform

esp32

### Platform Version(s)

1.1.0-1.1

aravindone, 2023-05-17:
@bzbarsky-apple @dhrishi @vivien-apple  Please comment on the error im facing

dhrishi, 2023-05-17:
@aravindone Please check if your host machine running the python utility is connected to the same network as the ESP32? The logs show that the python-repl could not find the ESP32 device on the operational network and hence the commissioning timed out after a while.

aravindone, 2023-05-17:
@dhrishi  im currently working in virtual ubuntu and the host was connected to the same network as ESP 32 network  and still the issue comes

bzbarsky-apple, 2023-05-17:
@aravindone At the point when the esp32 starts operational advertising (the `mDNS service published: _matter._tcp` bits), does doing `avahi-browse _matter._tcp -r` on the machine where you are running the Python repl see the esp32?

> After commissioning with google home sample app and if i want to connect through chip-tool again commission need to be happen ? how it should be done.

You have to get the Google Home sample app to open a commissioning window.

aravindone, 2023-05-28:
closing the ticket as the commission was successful in Direct ubuntu laptop

closed 2023-05-28 by aravindone

tunm1228, 2023-09-21:
[image]
For me, it's giving the same error

aravindone, 2023-09-21:
@tunm1228  can you share the log and proper screenshot from python-repl.

tunm1228, 2023-09-21:
I am using the command "./out/python_env/bin/chip-repl" and the error is as below
[image]

aravindone, 2023-09-21:
@tunm1228  have you tried with sudo with the command ?  and it says there is no config file in /tmp/repl-storage.json, you have to check that

tunm1228, 2023-09-21:
I don't know what path this file is in. Do you know how to create it?

aravindone, 2023-09-23:
@tunm1228 which commit are you using ?