project-chip/connectedhomeip #26491: [BUG] Can't build CHIP-Tool on Raspberry Pi

jjm202, 2023-05-10:
### Reproduction steps

On a Raspberry Pi (4GB RAM, 64GB), running TH v2.8

- Clone the connectedhomeip repo
- Follow the Linux ARM zap-tool instructions [here ](https://github.com/project-chip/connectedhomeip/blob/cca64af176705cf009d979826b5815ab56e152a7/docs/guides/BUILDING.md#installing-zap-tool)
- on the step "npm ci", we run into a series of failures (see attached)

[2023-05-10T21_17_36_682Z-debug-0.log](https://github.com/project-chip/connectedhomeip/files/11446156/2023-05-10T21_17_36_682Z-debug-0.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

cca64af176705cf009d979826b5815ab56e152a7

### Platform

raspi

### Anything else?

Perhaps additional documentation can be added for the correct version of "npm" so this can be avoided.

bzbarsky-apple, 2023-05-11:
@tecimovic @brdandu

jjm202, 2023-05-11:
If there is a simple way of cross-compiling on the x64, can you please list the steps? We are using WSL2 Ubuntu 20.04.

jjm202, 2023-05-11:
Better yet, is there a simple way to cross-compile chip-tool for arm64 using WSL2 Ubuntu 20.04? Can we list the steps here?

srisham, 2023-05-11:
You can update the toolchain/arm_gcc/ arm_toolchain.gni with your toolchain configurations such as CC, CXX etc, and run the following command to build:

- gn args out/custom
<< Add the following >>
target_os = "linux"
target_cpu = "arm"
custom_toolchain = "../toolchain/arm_gcc:arm_toolchain"

- Save the file
- ninja -C out/custom

jjm202, 2023-05-11:
So essentially something like 

& apt-get install gcc-arm-linux-gnueabihf

then

```
import("//build_overrides/build.gni")

import("${build_root}/toolchain/gcc_toolchain.gni")

template("arm_toolchain") {
  gcc_toolchain(target_name) {
    _tool_name_root = "arm-linux-gnueabihf-"
    ar = _tool_name_root + "ar"
    cc = _tool_name_root + "gcc"
    cxx = _tool_name_root + "g++"

    toolchain_args = {
      current_cpu = "arm"
      current_os = invoker.current_os
      is_clang = false

      forward_variables_from(invoker.toolchain_args, "*")
    }
  }
}
```

oidebrett, 2023-05-11:
Hi @jjm202 

It's not exactly what you're looking for but I got chip-tool built on RPi by pre generating the code from the zap file templates on my WSL2 Ubuntu 22.04. I then zipped up the zzz_pregenerated folder and copied to my RPi and unzipped. I then ran gn gen with the arg chip_code_pre_generated_directory pointing to the unzipped folder. And then continued to build the chip tool with ninja as normal. A bit cumbersome but it worked for me and I didn't want to install node on my RPi.

s07641069, 2023-05-12:
@jjm202
Also you can use docker image for build: "connectedhomeip/chip-build-crosscompile:latest"
./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-arm64-chip-tool-clang build"

jjm202, 2023-05-12:
Thanks everyone. The docker container build worked.

closed 2023-05-12 by jjm202