project-chip/connectedhomeip #26497: [BUG] flag is not supported by this version of the compiler: -Xlint:all

nmtoan91, 2023-05-11:
### Reproduction steps

Hi,
I'm building android chip-tool app on linux, I got this error
Hope someone can help me 
```
2023-05-11 16:04:16 INFO    [7/17] ACTION //src/controller/java:tlv__kotlinc(//build/toolchain/android:android_arm64)
2023-05-11 16:04:16 INFO    FAILED: gen/src/controller/java/tlv.classlist
2023-05-11 16:04:16 INFO    python3 ../../build/chip/java/kotlinc_runner.py --classdir obj/src/controller/java/classes --outfile gen/src/controller/java/tlv.classlist --build-config gen/src/controller/java/tlv.json -- -d obj/src/controller/java/classes @gen/src/controller/java/tlv.sources -Werror -Xlint:all -Xlint:deprecation
2023-05-11 16:04:16 INFO    warning: flag is not supported by this version of the compiler: -Xlint:all
2023-05-11 16:04:16 INFO    warning: flag is not supported by this version of the compiler: -Xlint:deprecation
2023-05-11 16:04:16 INFO    /home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../src/controller/java/src/chip/tlv/utils.kt:23:3: error: type inference failed. Expected type mismatch: inferred type is Int but Long was expected
2023-05-11 16:04:16 INFO      foldRightIndexed(0) { i, it, acc ->
2023-05-11 16:04:16 INFO      ^
2023-05-11 16:04:16 INFO    /home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../src/controller/java/src/chip/tlv/utils.kt:24:53: error: type mismatch: inferred type is Long but Int was expected
2023-05-11 16:04:16 INFO        (acc shl 8) or (if (i == lastIndex && isSigned) it.toLong() else (it.toLong() and 0xFF))
2023-05-11 16:04:16 INFO                                                        ^
2023-05-11 16:04:16 INFO    /home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../src/controller/java/src/chip/tlv/utils.kt:24:71: error: type mismatch: inferred type is Long but Int was expected
2023-05-11 16:04:16 INFO        (acc shl 8) or (if (i == lastIndex && isSigned) it.toLong() else (it.toLong() and 0xFF))
2023-05-11 16:04:16 INFO                                                                          ^
2023-05-11 16:04:16 INFO    Traceback (most recent call last):
2023-05-11 16:04:16 INFO      File "/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../build/chip/java/kotlinc_runner.py", line 146, in <module>
2023-05-11 16:04:16 INFO        sys.exit(main())
2023-05-11 16:04:16 INFO      File "/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../build/chip/java/kotlinc_runner.py", line 130, in main
2023-05-11 16:04:16 INFO        retcode = subprocess.check_call(kotlin_args + args.rest)
2023-05-11 16:04:16 INFO      File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
2023-05-11 16:04:16 INFO        raise CalledProcessError(retcode, cmd)
2023-05-11 16:04:16 INFO    subprocess.CalledProcessError: Command '['/usr/bin/kotlinc', '-classpath', 'lib/third_party/java_deps/protobuf-java-3.22.0.jar', '-d', 'obj/src/controller/java/classes', '@gen/src/controller/java/tlv.sources', '-Werror', '-Xlint:all', '-Xlint:deprecation']' returned non-zero exit status 1.
2023-05-11 16:04:16 INFO    ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/nuc17/Test/connectedhomeip/./scripts/build/build_examples.py", line 230, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/nuc17/Test/connectedhomeip/./scripts/build/build_examples.py", line 220, in cmd_build
    context.obj.Build()
  File "/home/nuc17/Test/connectedhomeip/scripts/build/build/__init__.py", line 71, in Build
    builder.build()
  File "/home/nuc17/Test/connectedhomeip/scripts/build/builders/builder.py", line 99, in build
    self._build()
  File "/home/nuc17/Test/connectedhomeip/scripts/build/builders/android.py", line 428, in _build
    self._Execute(
  File "/home/nuc17/Test/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/home/nuc17/Test/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['ninja', '-C', '/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool'] failed: 1
```

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

tetedoie, 2023-05-11:
In my experience, it seems that the `build_examples` script is skipping some code generation if folder/file already exist (and/or if the Gradle command(s) it runs are "UP-TO-DATE"). If you run it two times in row, the second run is way faster with lot less steps.  
Can you try deleting the `/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool` folder and re-run the `build_examples` script?

nmtoan91, 2023-05-11:
@tetedoie ,
I tried and got the same error

```
 [1359/1380] ACTION //src/controller/java:tlv__kotlinc(//build/toolchain/android:android_arm64)
2023-05-11 18:32:13 INFO    FAILED: gen/src/controller/java/tlv.classlist
2023-05-11 18:32:13 INFO    python3 ../../build/chip/java/kotlinc_runner.py --classdir obj/src/controller/java/classes --outfile gen/src/controller/java/tlv.classlist --build-config gen/src/controller/java/tlv.json -- -d obj/src/controller/java/classes @gen/src/controller/java/tlv.sources -Werror -Xlint:all -Xlint:deprecation
2023-05-11 18:32:13 INFO    warning: flag is not supported by this version of the compiler: -Xlint:all
2023-05-11 18:32:13 INFO    warning: flag is not supported by this version of the compiler: -Xlint:deprecation
2023-05-11 18:32:13 INFO    /home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../src/controller/java/src/chip/tlv/utils.kt:23:3: error: type inference failed. Expected type mismatch: inferred type is Int but Long was expected
2023-05-11 18:32:13 INFO      foldRightIndexed(0) { i, it, acc ->
2023-05-11 18:32:13 INFO      ^
2023-05-11 18:32:13 INFO    /home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../src/controller/java/src/chip/tlv/utils.kt:24:53: error: type mismatch: inferred type is Long but Int was expected
2023-05-11 18:32:13 INFO        (acc shl 8) or (if (i == lastIndex && isSigned) it.toLong() else (it.toLong() and 0xFF))2023-05-11 18:32:13 INFO                                                        ^
2023-05-11 18:32:13 INFO    /home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../src/controller/java/src/chip/tlv/utils.kt:24:71: error: type mismatch: inferred type is Long but Int was expected
2023-05-11 18:32:13 INFO        (acc shl 8) or (if (i == lastIndex && isSigned) it.toLong() else (it.toLong() and 0xFF))2023-05-11 18:32:13 INFO                                                                          ^
2023-05-11 18:32:13 INFO    Traceback (most recent call last):
2023-05-11 18:32:13 INFO      File "/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../build/chip/java/kotlinc_runner.py", line 146, in <module>
2023-05-11 18:32:13 INFO        sys.exit(main())
2023-05-11 18:32:13 INFO      File "/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../build/chip/java/kotlinc_runner.py", line 130, in main
2023-05-11 18:32:13 INFO        retcode = subprocess.check_call(kotlin_args + args.rest)
2023-05-11 18:32:13 INFO      File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
2023-05-11 18:32:13 INFO        raise CalledProcessError(retcode, cmd)
2023-05-11 18:32:13 INFO    subprocess.CalledProcessError: Command '['/usr/bin/kotlinc', '-classpath', 'lib/third_party/java_deps/protobuf-java-3.22.0.jar', '-d', 'obj/src/controller/java/classes', '@gen/src/controller/java/tlv.sources', '-Werror', '-Xlint:all', '-Xlint:deprecation']' returned non-zero exit status 1.
2023-05-11 18:32:13 INFO    [1360/1380] ACTION //:matter_build_venv._create_virtualenv(//third_party/pigweed/repo/pw_build/python_toolchain:python)
2023-05-11 18:32:14 INFO    [1361/1380] ld -shared lib/jni/arm64-v8a/libCHIPController.so
2023-05-11 18:32:14 INFO    ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/nuc17/Test/connectedhomeip/./scripts/build/build_examples.py", line 230, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/nuc17/Test/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/nuc17/Test/connectedhomeip/./scripts/build/build_examples.py", line 220, in cmd_build
    context.obj.Build()
  File "/home/nuc17/Test/connectedhomeip/scripts/build/build/__init__.py", line 71, in Build
    builder.build()
  File "/home/nuc17/Test/connectedhomeip/scripts/build/builders/builder.py", line 99, in build
    self._build()
  File "/home/nuc17/Test/connectedhomeip/scripts/build/builders/android.py", line 428, in _build
    self._Execute(
  File "/home/nuc17/Test/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/home/nuc17/Test/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['ninja', '-C', '/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool'] failed: 1
```

tetedoie, 2023-05-11:
On which commit/branch do you get this error ?

tetedoie, 2023-05-11:
So first, It's compiling properly on my side on the last commit of the main branch. I'm suspecting an issue with your Kotlin compiler. The error is about this Kotlin extension function https://github.com/project-chip/connectedhomeip/blob/bb3ce44a59c402b5cf4080f1013dbbe7272df5dc/src/controller/java/src/chip/tlv/utils.kt#L22 
While the error could make sense theoretically, (the `foldRightIndexed` initial value type 0(Int) (used to infer the return type) is different than the actual returned type (Long)), it's compiling on my side anyway and If I copy/paste the extension function into another project as is, it's working as well. 
Another thing I'm noticing, is that your script steps are slightly different than mine. 
You have 
```
[1359/1380] ACTION //src/controller/java:tlv__kotlinc(//build/toolchain/android:android_arm64)
[1360/1380] ACTION //:matter_build_venv._create_virtualenv(//third_party/pigweed/repo/pw_build/python_toolchain:python)
```
While I have
```
[1359/1380] stamp python/obj/matter_build_venv._generate_3p_requirements.stamp
[1360/1380] ACTION //:matter_build_venv._install_base_3p_deps(//third_party/pigweed/repo/pw_build/python_toolchain:python)
```

tetedoie, 2023-05-11:
Alternatively, you can try modifying the troubling extension function by replacing `0` by `0L` on this line https://github.com/project-chip/connectedhomeip/blob/bb3ce44a59c402b5cf4080f1013dbbe7272df5dc/src/controller/java/src/chip/tlv/utils.kt#L23 But you'll probably face other similar issue so we should identify the root cause instead.

nicelyjust, 2023-05-11:
please upgrade your kotlin version(1.7.+)

nmtoan91, 2023-05-11:
@nicelyjust,
I'm sorry, it this the latest version?
[image]

nicelyjust, 2023-05-11:
Still have this problem?

nmtoan91, 2023-05-11:
@nicelyjust thank you, I passed that step. But I got another error :D :D 

```
2023-05-11 22:12:04 INFO    [1372/1380] ACTION //src/controller/java:java__javac(//build/toolchain/android:android_arm64)
2023-05-11 22:12:04 INFO    FAILED: gen/src/controller/java/java.classlist
2023-05-11 22:12:04 INFO    python3 ../../build/chip/java/javac_runner.py --classdir obj/src/controller/java/classes --outfile gen/src/controller/java/java.classlist --build-config gen/src/controller/java/java.json -- -d obj/src/controller/java/classes @gen/src/controller/java/java.sources -Werror -Xlint:all -Xlint:deprecation -parameters -Xlint:-options -source 8 -target 8
2023-05-11 22:12:04 INFO    ../../src/controller/java/zap-generated/chip/devicecontroller/ChipClusters.java:124: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO        protected void finalize() throws Throwable {
2023-05-11 22:12:04 INFO                       ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/zap-generated/chip/devicecontroller/ChipClusters.java:125: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO          super.finalize();
2023-05-11 22:12:04 INFO               ^
2023-05-11 22:12:04 INFO    error: warnings found and -Werror specified
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/GetConnectedDeviceCallbackJni.java:40: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO      protected void finalize() throws Throwable {
2023-05-11 22:12:04 INFO                     ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/GetConnectedDeviceCallbackJni.java:41: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO        super.finalize();
2023-05-11 22:12:04 INFO             ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/ChipDeviceController.java:1085: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO      protected void finalize() throws Throwable {
2023-05-11 22:12:04 INFO                     ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/ChipDeviceController.java:1086: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO        super.finalize();
2023-05-11 22:12:04 INFO             ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/InvokeCallbackJni.java:40: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO      protected void finalize() throws Throwable {
2023-05-11 22:12:04 INFO                     ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/InvokeCallbackJni.java:41: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO        super.finalize();
2023-05-11 22:12:04 INFO             ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/ReportCallbackJni.java:53: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO      protected void finalize() throws Throwable {
2023-05-11 22:12:04 INFO                     ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/ReportCallbackJni.java:54: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO        super.finalize();
2023-05-11 22:12:04 INFO             ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/ReportEventCallbackJni.java:53: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO      protected void finalize() throws Throwable {
2023-05-11 22:12:04 INFO                     ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/ReportEventCallbackJni.java:54: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO        super.finalize();
2023-05-11 22:12:04 INFO             ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/WriteAttributesCallbackJni.java:40: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO      protected void finalize() throws Throwable {
2023-05-11 22:12:04 INFO                     ^
2023-05-11 22:12:04 INFO    ../../src/controller/java/src/chip/devicecontroller/WriteAttributesCallbackJni.java:41: warning: [removal] finalize() in Object has been deprecated and marked for removal
2023-05-11 22:12:04 INFO        super.finalize();
2023-05-11 22:12:04 INFO             ^
2023-05-11 22:12:04 INFO    1 error
2023-05-11 22:12:04 INFO    14 warnings
2023-05-11 22:12:04 INFO    Traceback (most recent call last):
2023-05-11 22:12:04 INFO      File "/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../build/chip/java/javac_runner.py", line 146, in <module>
2023-05-11 22:12:04 INFO        sys.exit(main())
2023-05-11 22:12:04 INFO      File "/home/nuc17/Test/connectedhomeip/out/android-arm64-chip-tool/../../build/chip/java/javac_runner.py", line 130, in main
[... cut ...]
2023-05-11 22:12:04 INFO    subprocess.CalledProcessError: Command '['/usr/bin/javac', '-classpath', 'lib/third_party/java_deps/protobuf-java-3.22.0.jar:lib/src/controller/java/android.jar:lib/third_party/java_deps/jsr305-3.0.2.jar:lib/src/controller/java/libCHIPTlv.jar', '-d', 'obj/src/controller/java/classes', '@gen/src/controller/java/java.sources', '-Werror', '-Xlint:all', '-Xlint:deprecation', '-parameters', '-Xlint:-options', '-source', '8', '-target', '8']' returned non-zero exit status 1.
```

nmtoan91, 2023-05-15:
@tetedoie I'm sorry to bother you again. But can you take a look at the log and suggest to me what should I do, please. 
By the way, it is very convenient if you can send the chip-tool apk file, I just need the android app for testing.

tetedoie, 2023-05-15:
Hi @nmtoan91, your last issue is a problem with the Java compiler. The `Object.finalize()` method is deprecated since Java 9 and the command have the `-Werror` parameter (which would raise an error if any warning is present). Each of the `finalize()` methods in the code is annotated with `@SuppressWarning("deprecated")`, so those warnings (and the error) should go away but it seems that your Java compiler ignore the annotation. I'm pretty sure I faced this issue in the past but I don't remember in which context and how to fix it. In the meantime here is the [APK](https://drive.google.com/file/d/10AIZSwBjtwkgn80s1-93jXhuwipFc4tF/view?usp=share_link) (It's for ARM64 on commit a33249db16d3bb351038e68d997db4c0ba5d003c). Please let me know when you downloaded it so I can delete it.

Rajesh0071998, 2023-06-29:
Hi @tetedoie,  as i am also facing the same issues with the java compiler. Is there any way to solve this and your APK link is also expired. so if you can get the APK link then it would be great or how can we solve this issue.

kiss2sky, 2024-12-13:
1.3-branch
FAILED: gen/src/controller/java/java.classlist

i am also facing the same issues,how can we solve this issue.

yunhanw-google, 2025-02-05:
We have upgraded Gradle version 7.3.3 and JDK version 17.0 for android build,  you can refer this PR for latest instructions, https://github.com/project-chip/connectedhomeip/pull/37368/files, if there is still problem, feel free to create a new issue, thanks

closed 2025-02-05 by yunhanw-google