project-chip/connectedhomeip #26509: [SQA] ClusterRevision Attribute value missmatch

Survensa, 2023-05-11:
**Problem**
As per Spec, ClusterRevision attribute value has to be updated for below mentioned clusters
- [ ] Color Control Cluster
- [ ] Descriptor Cluster
- [ ] Bridged Device Basic Information Cluster
- [ ] Basic Information Cluster
- [ ] Thermostat Cluster

**For example:**
> In  Color Control Cluster
> **Expected behavior**: ClusterRevision value as 6
> 
> **Actual behavior** : ClusterRevision value as 5
> 
> **Steps to reproduce**
> ./chip-tool colorcontrol read cluster-revision 1 1
> **Log**
> [TC-CC-1.1.txt](https://github.com/project-chip/connectedhomeip/files/11452341/TC-CC-1.1.txt)

**System configuration:**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : Onnetwork
Commit-ID:6c0c5ea6d2b8f38c6b36b83195b6ee55384bb599
Dut application: all-clusters-app

**Spec Reference:**
https://github.com/CHIP-Specifications/connectedhomeip-spec/tree/master/src/

bzbarsky-apple, 2023-05-11:
This issue is not really actionable as filed.  ClusterRevision got changed on those clusters because their behaviors changed.  For each cluster, that behavior change needs to be implemented, so there should be separate issues per-cluster.

Note that for some clusters those issues already exist.

cjandhyala, 2023-05-16:
closing this ticket as we are tracking this with individual tickets..

closed 2023-05-16 by cjandhyala