project-chip/connectedhomeip #26512: [SQA] Issue with Ble pairing.

Saravana-kr22, 2023-05-11:
### Description:
There is an issue in ble pairing, While chip-tool tries to pair using ble -wifi or ble-thread chip-tool is struct at the moment it finds the device as below.

> [1683804286.845358][20060:20061] CHIP:BLE: Device 0B:8F:44:BE:EF:DC does not look like a CHIP device.
> [1683804286.857739][20060:20061] CHIP:BLE: New device scanned: C9:37:E9:D1:9A:6F
> [1683804286.857841][20060:20061] CHIP:BLE: Device discriminator match. Attempting to connect.

**Steps to reproduce:**  `./chip-tool pairing ble-wifi 1 <ssid> <password> 20202021 3840 `

**System configuration:**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM and Thread - nrf52840-dk.
Platform : ble-wifi,ble-threda
Commit-ID:6c0c5ea6d2b8f38c6b36b83195b6ee55384bb599
Dut application: all-clusters-app

**Logs**:
ble-wifi:
[Wifi log TH.txt](https://github.com/project-chip/connectedhomeip/files/11453264/Wifi.log.TH.txt)
[Wifi log DUT.txt](https://github.com/project-chip/connectedhomeip/files/11453265/Wifi.log.DUT.txt)
ble-thread:
[threadlog.txt](https://github.com/project-chip/connectedhomeip/files/11453286/threadlog.txt)

bzbarsky-apple, 2023-05-11:
@tehampson

bzbarsky-apple, 2023-05-11:
Chances are https://github.com/project-chip/connectedhomeip/pull/26418 fixed this.

oidebrett, 2023-05-11:
> @oidebrett Since you seems to be using a tag, that doesn't have the PR that is breaking I think that your issue might be different. There has been quite a few changes around how BLE is done, as well as strict checks that you are in the matter context. If you are not able to reproduce on tip of tree master branch then I don't know if I can help out for a particular tag

Now works! I possibly was using a more recent commit. Anyway I tried the commit hash for the above fix (d55c592eb3a960e6ca8f4b3926d05d76665f89f3) and commissioning over BLE now works for me. I must have been mistaken on the commit hash I was using when I experienced the error.

[1683833109.175347][1094:1096] CHIP:TOO: Device commissioning completed with success

Thanks!