project-chip/connectedhomeip #26515: [Documentation] Guidelines to add customer clusters to a Matter-enabled device

shansong99, 2023-05-11:
### Documentation issues

Hi, I would like to check with Matter experts whether this is possible:

- As Matter only provides limited number of clusters mainly for home automation, is it possible to add some "customer clusters" by myself into the Matter-enabled product?
- Will such product compliant with Matter standard and can get the certification? (all other Matter mandatory clusters will be correctly implemented in this product).
- on top if this, is it possible to add additional security controls on such customer cluster so that only phone APP with addtional certificaiton can access it?

It'll be nice if there is a guideline document to show how to add such customer cluster in Matter.

Thanks for your help!

bzbarsky-apple, 2023-05-11:
> is it possible to add some "customer clusters" by myself into the Matter-enabled product

Yes, you can add manufacturer-specific clusters, as long as their cluster ids are in the appropriate range for your vendor id.

> is it possible to add additional security controls on such customer cluster so that only phone APP with addtional certificaiton can access it

You mean outside of just the normal ACL setup?

shansong99, 2023-05-11:
> Yes, you can add manufacturer-specific clusters, as long as their cluster ids are in the appropriate range for your vendor id.
>
>
> You mean outside of just the normal ACL setup?

Yes, I mean out side of the normal ACL setup, is it possible to put addtiona constraints on a customer cluster so that only authorized app can access it.