project-chip/connectedhomeip #26554: [XML] Update Identify Cluster XML to match updated spec

ReneJosefsen, 2023-05-12:
The data types names has been updated in Identify Cluster and must be updated in the XML as well.

**Added:**
IdentifyTypeEnum
EffectIdentifierEnum
EffectVariantEnum

Spec PR: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6470 (already merged)

closed 2023-05-26 by bzbarsky-apple

hccqqa, 2023-08-16:
Hi ReneJosefsen, 

The Link is can't to use , please see the below : 
Spec PR: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6470 (already merged)

and we have one question to ask, what can i do using zap tool to generate the below identifial clusters types , please see the below :

IdentifyTypeEnum
EffectIdentifierEnum
EffectVariantEnum

ReneJosefsen, 2023-08-16:
Not sure I completely understand but, as far as I know these should be automatically generated when you run the gode gen step in the build process?

hccqqa, 2023-08-17:
Hi ReneJosefsen,
     Thanks for your quick reply,
      May I ask one more question?
      could you let me know what do you mean the "gode gen"?
      it is "code generate"?

ReneJosefsen, 2023-08-17:
No problem 😄 

Yes, one of the steps in the build process is to generate code from the .zap and .matter file.