project-chip/connectedhomeip #26568: [BUG] Repeating Commission in chip-repl will fail

yzm157, 2023-05-15:
### Reproduction steps

1. Execute chip-all-clusters-app on the Raspberry Pi
`sudo rm -rf /tmp/chip_* && sudo ./chip-all-clusters-app --wifi`
2. Run chip-repl on ubuntu, and CommissionWiFi
```
chip-repl -d
devCtrl.CommissionWiFi(3840, 20202021, 1, 'rpi', 'rpimatter')    # Commission Succeed
```
3. Stop chip-all-clusters-app, and run again
`[Ctrl + C] && sudo rm -rf /tmp/chip_* && sudo ./chip-all-clusters-app --wifi`
4. CommissionWiFi again in the chip-repl in step 2
`devCtrl.CommissionWiFi(3840, 20202021, 1, 'rpi', 'rpimatter')    # Commission Failed`

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

4083b32732b04aeec63b8f03ebd1e12ade78dd0f

### Platform

python

bzbarsky-apple, 2023-05-15:
I assume the issue is that you are trying to commission with the same node id that was used before?

yzm157, 2023-05-16:
> I assume the issue is that you are trying to commission with the same node id that was used before?

Yes, after resetting the hardware and recommissioning, all parameters have not changed