project-chip/connectedhomeip #26571: [ACT-1.1] EventList verification failure.

manjunath-grl, 2023-05-15:
### Reproduction steps

Summary:
  As per spec and testplan `ACTIONS` cluster  StateChanged and ActionFailed events are mandatory but when we verified it using YAML got 0 entries.
Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACT_EventList_Failure

Steps to reproduce:
1. Enable the enable_eventlist_attribute to true
    `gn gen out/host --args='enable_eventlist_attribute=true'`
1 .Run zap
    `./scripts/tools/zap_regen_all.py --type tests`
3. Build
    `./scripts/run_in_build_env.sh \
                     "./scripts/build/build_examples.py \
                        --target linux-x64-chip-tool-no-ble-tsan-clang \
                        --target linux-x64-all-clusters-no-ble-tsan-clang \
                        --target linux-x64-lock-no-ble-tsan-clang \
                        --target linux-x64-ota-provider-no-ble-tsan-clang \
                        --target linux-x64-ota-requestor-no-ble-tsan-clang \
                        --target linux-x64-tv-app-no-ble-tsan-clang \
                        --target linux-x64-bridge-no-ble-tsan-clang \
                        build \
                     "`
4. Run YAML scripts
    `./scripts/run_in_build_env.sh \
                  "./scripts/tests/run_test_suite.py \
                     --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool \
                     run \
                     --iterations 1 \
                     --all-clusters-app ./out/linux-x64-all-clusters-no-ble-tsan-clang/chip-all-clusters-app \
                     --lock-app ./out/linux-x64-lock-no-ble-tsan-clang/chip-lock-app \
                     --ota-provider-app ./out/linux-x64-ota-provider-no-ble-tsan-clang/chip-ota-provider-app \
                     --ota-requestor-app ./out/linux-x64-ota-requestor-no-ble-tsan-clang/chip-ota-requestor-app \
                     --tv-app ./out/linux-x64-tv-app-no-ble-tsan-clang/chip-tv-app \
                     --bridge-app ./out/linux-x64-bridge-no-ble-tsan-clang/chip-bridge-app \
                  "`

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

a33249db16d3bb351038e68d997db4c0ba5d003c

### Platform

other

### Anything else?

Spec screenshot:
[image]

Execution logs:
[image]
[execution_Log.txt](https://github.com/project-chip/connectedhomeip/files/11476635/execution_Log.txt)

bzbarsky-apple, 2023-05-15:
all-clusters-app does not implement Actions cluster per spec and really cannot.  You need to be testing actions cluster bits against bridge-app, @manjunath-grl .  That probably means adding the relevant condition to https://github.com/project-chip/connectedhomeip/blob/b4fb147dc47ec6bb95b174f3b8fdebf7f2088891/scripts/tests/chiptest/__init__.py#L169-L170

That said, as far as I can tell, bridge-app also does not claim to expose these events, even though it does seem to implement them.  @turon, do you know why?

turon, 2023-05-15:
> That said, as far as I can tell, bridge-app also does not claim to expose these events, even though it does seem to implement them. @turon, do you know why?

I'm not sure why the bridge app wouldn't expose these Actions events it implements.  Perhaps it was simply an oversight when the implementation was added?

cjandhyala, 2023-05-16:
@manjunath-grl Can you close this ticket per conversation with suraj and sankar on using bridge app for your testing.

closed 2023-05-22 by bzbarsky-apple