project-chip/connectedhomeip #26582: Generate sample revoked DACs and PAI certs for 0xFFF1 vendor ID

tcarmelveilleux, 2023-05-15:
Need to generate CRLs for revocation testing.

- 3x DACs with different serial numbers for VID 0xFFF1, PID 0x8001
  - Signing PAI is `./credentials/development/attestation/Matter-Development-PAI-FFF1-noPID-Cert.pem` and `/credentials/development/attestation/Matter-Development-PAI-FFF1-noPID-Key.pem`
  - Issue the DACs with that PAI
  - Revoke them
  - Create a CRL for the PAI that includes the 3 revoked DACs

- Create 1 new PAI with the test VID=FFF1 PAA (`./credentials/development/paa-root-certs/Chip-Test-PAA-FFF1-Cert.pem` and `./credentials/test/attestation/Chip-Test-PAA-FFF1-Key.pem`
  - Issue a DAC with that PAI for PID 0x8001
  - Revoke the PAI
  - Create a CRL for the PAA that includes that new PAI's cert

The CRLs will need to be hosted permanently in a CDN. The CRLs need to list a reasonably long NextUpdate time.

tcarmelveilleux, 2023-11-27:
FYI @CBonnell

shubhamdp, 2024-12-16:
Also add a set of CRLs and certs for the indirect CRL signing use cases.

Please see https://github.com/project-chip/connectedhomeip/pull/36838#pullrequestreview-2502738010 for more details

shubhamdp, 2024-12-23:
Checklist:
- [ ] CRL signed using PAA delegated CRL signer
- [ ] CRL signed using PAI delegated CRL signer
- [ ] CRL with Issuing Distribution Point CRL extension with the matching GeneralName URI (https://github.com/project-chip/connectedhomeip/pull/36838#discussion_r1884221926)
- [ ] CRL with Certificate Issuer CRL entry extension, each entry has the extension
- [ ] CRL with Certificate Issuer CRL entry extension, only first entry has the extension

PR #36929 (2025-03-28): credentials: sample delegated crl signer certs and CRLs
Generated the samples as described in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/10308

Added README file which describes the content of the directory.

Its a follow up on #36838, and ticks all items from https://github.com/project-chip/connectedhomeip/issues/26582#issuecomment-2558952413.

Fixes #26582

Generated the delegated CRL signers using the cryptography module from python.

### Testing
- This PR is the building block for automating the DA revocation implementation. It adds the test data for generating the  [... cut]
closed 2025-03-28 by mergify[bot]