project-chip/connectedhomeip #26591: [BUG] [ESP32-C3] ESP-IDF v4.4.4 cannot build lighting-app on MatterSDK v1.1.0.1

mmt-10, 2023-05-16:
### Reproduction steps

I followed [https://github.com/project-chip/connectedhomeip/blob/v1.1.0.1/docs/guides/esp32/setup_idf_chip.md](https://github.com/project-chip/connectedhomeip/blob/v1.1.0.1/docs/guides/esp32/setup_idf_chip.md)

```bash
$ cd ${IDF_PATH}
$ source export.sh
$ cd ${MATTER_SDK}
$ source scripts/activate.sh
$ cd examples/lighting-app/esp32
$ idf.py set-target esp32c3
$ idf.py build
```
Then, I got below message.
[idf-v4.4.4.log](https://github.com/project-chip/connectedhomeip/files/11483408/idf-v4.4.4.log)

I retried `install.sh`, but `idf.py build` was failed same as above log.
`install.sh` log is here.
[idf-v4.4.4_install.log](https://github.com/project-chip/connectedhomeip/files/11483466/idf-v4.4.4_install.log)

```bash
$ cd ${IDF_PATH}
$ git log -1
commit e8bdaf91986a41678adc6be13888fc037b1acb68 (HEAD, tag: v4.4.4)
Author: Xiao Xufeng <xiaoxufeng@espressif.com>
Date:   Fri Jan 20 02:12:49 2023 +0800
```

Host OS:Ubuntu 22.04

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 8f66f4215bc0708efc8cc73bda80620e67d8955f (HEAD, tag: v1.1.0.1)

### Platform

esp32

bzbarsky-apple, 2023-05-16:
@shubhamdp @dhrishi

shubhamdp, 2023-05-17:
> bitstring>=3.1.6,<4
To install the missing packages, please run "/home/mmt-10/matter/tools/esp-idf-v4.4.4/install.sh"

Also, I can see the bistring is already part of [requirements.esp32.txt](https://github.com/project-chip/connectedhomeip/blob/v1.1.0.1/scripts/setup/requirements.esp32.txt#L7), there could be possibly two problems. 1. Its not installed or 2. Version mismatch.

First can be fixed running bootstrapping again
```
source scripts/bootstrap.sh
```

And for second let's check the bitstring version, I just wan't to make sure that it matches the above constraint.
```
source scripts/activate.sh
python3 -m pip show bitstring
```

I think, it would be nice to have a version constraint in the requirements file. Will fix that.

mmt-10, 2023-05-17:
Thank you for your reply.
`bitstring` version was `4.0.1`.

I specified a version of bitstring and installed it.
It resolved.
```bash
$ python3 -m pip install bitstring==3.1.9
```

BTW,
How about below error?
> I tried ESP-IDF v5.0.
> And, it has error.
>
>>    /home/mmt-10/matter/matter-v1.1.0.1/.environment/pigweed-venv/bin/python: No module named esptool

shubhamdp, 2023-05-18:
Ohh sorry, I may have missed the below error
> /home/mmt-10/matter/matter-v1.1.0.1/.environment/pigweed-venv/bin/python: No module named esptool

In IDF v4.x `esptool.py` is part of repository but in IDF v5.x its moved out of repository and available as the python package. Please try to install it..
```
python3 -m pip install esptool
```

And do not close the issue, I'll go through these exact scenarios and try to compare the documentation guide and fix them.

mmt-10, 2023-05-18:
Thank you for your reply.

I tried `python3 -m pip install esptool` with IDF v5.0.
It resolved.

However, I got other error as bleow.

[idf-v5.0-20230518.log](https://github.com/project-chip/connectedhomeip/files/11504262/idf-v5.0-20230518.log)

> [1004/1008] Generating ld/sections.ldFAILED: esp-idf/esp_system/ld/sections.ld /home/mmt-10/matter/matter-v1.1.0.1/examples/lighting-app/esp32/build/esp-idf/esp_system/ld/sections.ld

> Traceback (most recent call last):
>   File "/home/mmt-10/esp32/esp-idf-v5.0/tools/ldgen/ldgen.py", line 16, in <module>
>     from ldgen.entity import EntityDB
>   File "/home/mmt-10/esp32/esp-idf-v5.0/tools/ldgen/ldgen/entity.py", line 12, in <module>
>     from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
> ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/home/mmt-10/matter/matter-v1.1.0.1/.environment/pigweed-venv/lib/python3.10/site-packages/pyparsing.py)
> ninja: build stopped: subcommand failed.

shubhamdp, 2023-05-18:
There is some version conflict between pyparsing package in IDF v4.4.x and V5.x  :-(

Can you try to uninstall the current pyparsing package and install the newer version.
```
python3 -m pip uninstall -y pyparsing && python3 -m pip install pyparsing
```

But, when you start using IDF v4.4 you will have to do the same again to downgrade the pyparsing version
```
python3 -m pip uninstall -y pyparsing && python3 -m pip install pyparsing==2.3.0
```

closed 2023-05-19 by mmt-10

404-code, 2023-11-27:
> There is some version conflict between pyparsing package in IDF v4.4.x and V5.x :-(
>
> Can you try to uninstall the current pyparsing package and install the newer version.
>
> ```
> python3 -m pip [...]

hello @shubhamdp ,I want to know if i use the idf v5.1 to build matter project, is pyparsing versiong also required to meet the condition: pyparsing==2.3.0?