project-chip/connectedhomeip #26637: ASR platform manager synchronization seems to be completely broken

bzbarsky-apple, 2023-05-17:
See https://github.com/project-chip/connectedhomeip/pull/26361#pullrequestreview-1431308530

@tx2rx @andy31415

bzbarsky-apple, 2023-05-17:
Note that I only reviewed PlatformManager and TimeSupport from https://github.com/project-chip/connectedhomeip/pull/26361 and both were broken.  I have extremely low confidence in any of the rest of it...

tx2rx, 2023-05-17:
> Also, where is _TryLockChipStack implemented for ASR? I don't see it.

Because in src/ , that is not used before. Just in case ,we will add it soon later

tx2rx, 2023-05-17:
src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp may cause the same problem.

tx2rx, 2023-05-18:
In addition, src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp may cause the same problem:

        BaseType_t eventReceived = pdFALSE;
        {
            // Unlock the CHIP stack, allowing other threads to enter CHIP while
            // the event loop thread is sleeping.
            StackUnlock unlock;
            eventReceived = xQueueReceive(mChipEventQueue, &event, waitTime);
**StackLock lock;** // should it open lock again? @bzbarsky-apple 
        }

tx2rx, 2023-05-18:
refer https://github.com/project-chip/connectedhomeip/pull/19470

bzbarsky-apple, 2023-05-18:
> // should it open lock again?

It does, in the destructor of the StackLock.

PR #26645 (2023-05-24): fix asr platform issue
fix #26637 

1,modify PlatformManagerImpl implementation
2,bump third_party/asr/xxx/asr_sdk
3,remove unless code in SystemTimeSupport
closed 2023-05-24 by bzbarsky-apple