project-chip/connectedhomeip #26639: [chiptool.py] Failed to run test cases TC-WNCV-3.4 , TC- WNCV-3.5 and TC-WNCV-4.1

ccruzagralopes, 2023-05-17:
### Reproduction steps / Feature

When trying to run test cases `TC-WNCV-3.4` , `TC- WNCV-3.5` and `TC-WNCV-4.1` using `chiptool.py`, we get the following failure during a step with command `WaitForMs`:
```
(...)
File "/root/scripts/py_matter_yamltests/matter_yamltests/pseudo_clusters/clusters/delay_commands.py", line 61, in WaitForMs
time.sleep(duration_in_ms / 1000)
TypeError: unsupported operand type(s) for /: 'str' and 'int'
```

Example of yaml step that is failing (`TC-WNCV-3.4`):
```yaml
fastMotionDuration:
    type: int16u
    defaultValue: 3000

(...)

- label:
        "1b: TH Waits for fastMotionDuration seconds movement(s) on the device"
    cluster: "DelayCommands"
    command: "WaitForMs"
    arguments:
        values:
            - name: "ms"
            value: fastMotionDuration
```

### Platform

other

### Type

Manually tested with SDK

### Anything else?

Related issues:
- Issue filed by QA: https://github.com/CHIP-Specifications/chip-certification-tool/issues/822
- Previous issue that seems to have the same error and has been fixed: https://github.com/project-chip/connectedhomeip/issues/26461

bzbarsky-apple, 2023-05-17:
@ccruzagralopes This was fixed a few days ago in https://github.com/project-chip/connectedhomeip/pull/26541, as you note.  What SHA are you using?  It looks like you are using something that predates that fix.

ccruzagralopes, 2023-05-17:
> @ccruzagralopes This was fixed a few days ago in #26541, as you note. What SHA are you using? It looks like you are using something that predates that fix.

Oh, I didn't realize that the fix also worked for this command, sorry.

The SHA I'm using is `226a5772cf0d1385dcccbffaeba6c5fa96844d7b`

ccruzagralopes, 2023-05-17:
I just ran these tests with the latest SDK SHA and they're all passing

closed 2023-05-17 by ccruzagralopes

bzbarsky-apple, 2023-05-17:
Thank you for the quick followup!