project-chip/connectedhomeip #26646: [BUG] Modifications to temperature control cluster as per the spec changes

dhrishi, 2023-05-18:
### Reproduction steps

NA

### Bug prevalence

NA

### GitHub hash of the SDK that was being used

master

### Platform

core

### Anything else?

The spec issue https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7040 would require changes in the SDK code. Need to make the modifications as and when the spec changes converge

dhrishi, 2023-05-18:
@jadhavrohit924 Please track this

dhrishi, 2023-05-18:
@bzbarsky-apple As discussed, requested this to be a cert blocker. Will you add the `cert blocker` label? Or am I allowed to add?

cjandhyala, 2023-05-18:
I think its related to  #26601 which is already filed

dhrishi, 2023-05-18:
@cjandhyala The above seems to be specific to the thermostat cluster. This is about the temperature control cluster

jadhavrohit924, 2023-08-22:
Fixed in https://github.com/project-chip/connectedhomeip/commit/701fcb21f9a

closed 2023-08-22 by jadhavrohit924