project-chip/connectedhomeip #26647: [chip-tool]can I change the save path for chip_tool_config.ini?

mmt-10, 2023-05-18:
### Reproduction steps

I would like to keep list of commissioned clusters.
After reboot linux, then chip-tool forgets clusters.

[https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool#using-the-client-to-commission-a-device](https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool#using-the-client-to-commission-a-device)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 8f66f4215bc0708efc8cc73bda80620e67d8955f (HEAD, tag: v1.1.0.1)

### Platform

other

bzbarsky-apple, 2023-05-18:
@mmt-10 On the revision you're on, you can set TMPDIR (while running chip-tool) to the directory where you want chip-tool to place its data.

On tip there's a `--storage-directory` argument you can use.

bzbarsky-apple, 2023-05-18:
But perhaps we should update the documentation to mention `--storage-directory`...

closed 2023-05-18 by bzbarsky-apple

mmt-10, 2023-05-19:
Thank you for your reply.
I checked v1.1.0.1 does not support  `--storage-directory`.
but, master supports `--storage-directory`.

bzbarsky-apple, 2023-05-19:
@mmt-10 Yes, that's what I said above, right?  On 1.1.0.1 you can set the TMPDIR environment variable; on master you can do that or use the new command-line option.