project-chip/connectedhomeip #26671: src/setup_payload/python/generate_setup_payload.py

KuanHug, 2023-05-19:
### Documentation issues

spec  v1.0 Chapter 5.1.3.1. I see payload format that is "Version"、"VendorID"、"ProductID"...

in generate_setup_payload.py,The first is "Padding". Could somebody tell me why?

shubhamdp, 2023-05-19:
@KuanHug If you see here https://github.com/project-chip/connectedhomeip/blob/e75e729eb9ce78abf439c8bb332af01ba560d542/src/setup_payload/python/generate_setup_payload.py#L110-L118 and compare it with spec the order is reversed. Thats why padding is at the very top in code and at the bottom in spec.

KuanHug, 2023-05-22:
@shubhamdp Oh okay , I understand thanks.

Another problem, I don't understand this function to do qrcode string.
https://github.com/project-chip/connectedhomeip/blob/bfa05804ac3c224c9e70921a857ffcac32ccc7f4/src/setup_payload/python/Base38.py#L44-L47

for example,
[image]
But the result is "base38[W,2,7,K,2]" with the base32.py function ,is not match. So I don't know why.