project-chip/connectedhomeip #26677: [Feature] Ble Advertising Feature Implementation for Darwin

ejunu, 2023-05-19:
### Feature description

@vivien-apple 
Hello, I would like to request the addition of a new feature to the Matter Darwin.

We are currently working on integrating Matter with Thread IoT devices and have developed an emulator called ioter for Linux (repository: [https://github.com/Samsung/ioter](https://github.com/Samsung/ioter)). As we expand our development to support Mac environments, we came across your commit regarding the Ble scan functionality in the Matter Darwin codebase. We believe your expertise could be instrumental in implementing a Ble advertising feature for Matter Darwin on Mac.

We request assistance in adding the Ble advertising functionality to Matter Darwin for Mac. This feature would not only benefit our project but also prove valuable to other developers working on Mac platforms.

Thank you. And I look forward to your feedback on this issue.

### Platform

darwin

junyong-sim, 2023-05-22:
I tried to implement BLE advertising with CBPeripheralManager::startAdvertising API. (https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393252-startadvertising)

With that, advertising was working, but, I wasn't able to add service data at an advertising packet.
It seems that IOS APIs only supports `CBAdvertisementDataLocalNameKey `and `CBAdvertisementDataServiceUUIDsKey`.
I don't know if it was because of no service data, so BLE connection was not established between controller and ios things.

woody-apple, 2023-06-23:
Sadly this is not something that Matter can do without further support from new Apple API. Please file a feedback assistant ticket on Home as follows, and reference this GitHub issue:

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

Please also note the FBA number here, and we can follow up.

closed 2023-06-23 by woody-apple