project-chip/connectedhomeip #26684: [Feature]

LucaFagan, 2023-05-19:
### Feature description

I'm using 

`      let parser = MTRQRCodeSetupPayloadParser(base38Representation: "MT:5KR00-Q000KA0648G00")

        if let payload = try? parser.populatePayload()
        {
            let serialNumber = payload.serialNumber
            let productIdHex = String(format:"%04X", payload.productID.uint64Value)
            let vendorIdHex = String(format:"%04X", payload.vendorID.uint64Value)
            let versionHex = String(format:"%02X", payload.version.uint64Value)
        }
        `
to parse a commissioning QR code but serialNumber is nil.

Is it possible to have serialNumber in the QR code string ? How can generate such a QR code ?

### Platform

darwin

bzbarsky-apple, 2023-05-19:
The serial number would need to live in the TLV extension bits of the payload.  Specifically, you would have an anonymous-tagged structure with a single field: context-specific tag 0, type string or unsigned integer, value serial number.  So something like hex bytes `152c0002424318` for a two-char string "BC".

```
% chip-tool payload generate-qrcode --existing-payload "MT:5KR00-Q000KA0648G00" --tlvBytes hex:152c0002424318 --version 0
[1684509906518] [59255:25761244] [TOO] QR Code: MT:4KR00-Q000KA064IJ3P0GUE20A6.S0

```
to generate a payload like the one you have above but with valid version number and those TLV bytes.  That leads to:
```
% chip-tool payload parse-setup-payload MT:4KR00-Q000KA064IJ3P0GUE20A6.S0
[1684509957942] [59353:25762171] [SPL] Parsing base38Representation: MT:4KR00-Q000KA064IJ3P0GUE20A6.S0
[1684509957942] [59353:25762171] [SPL] Version:             0
[1684509957942] [59353:25762171] [SPL] VendorID:            4969
[1684509957942] [59353:25762171] [SPL] ProductID:           32768
[1684509957942] [59353:25762171] [SPL] Custom flow:         0    (STANDARD)
[1684509957942] [59353:25762171] [SPL] Discovery Bitmask:   0x00 (NONE)
[1684509957942] [59353:25762171] [SPL] Long discriminator:  3840   (0xf00)
[1684509957942] [59353:25762171] [SPL] Passcode:            20202021
[1684509957943] [59353:25762171] [SPL] SerialNumber:        BC
```

I should fix `MTRSetupPayload`'s `qrCodeString` to encode the `serialNumber` in there if it's set....

Also, you should probably use `MTRSetupPayload.setupPayload(onboardingPayload: ...)` for parsing, not the deprecated `MTRQRCodeSetupPayloadParser` bit.

closed 2023-05-19 by bzbarsky-apple

LucaFagan, 2023-05-20:
It works, thanks