project-chip/connectedhomeip #26715: java.lang.IllegalStateException CHIP Error 0x00000003: Incorrect state: 3

sahityadubey, 2023-05-22:
### Reproduction steps

I am trying to use invoke method for turning on/off matter devices from the Android app but getting the below error.

Error:
 _java.lang.IllegalStateException: ../../src/lib/core/TLVReader.cpp:532: CHIP Error 0x00000003: Incorrect state: 3_

Code:
```
  val jsonObject = JSONObject()
  jsonObject.put("type", "Structure")
  jsonObject.put("value", "")

  val tlvWriter = TlvWriter()

  val invokeElement =  InvokeElement.newInstance(endpoint: 1, clusterId: 257, commandId: 0, tlvWriter.getEncoded(), jsonObject.toString())
  val result = chipClient.invoke(chipClient.getConnectedDevicePointer(nodeId), invokeElement, 1000)

ChipClient ->
suspend fun invoke(
      devicePtr: Long,
      invokeElement: InvokeElement,
      timedRequestTimeoutMs: Int = DEFAULT_TIMEOUT,
      imTimeoutMs: Int = DEFAULT_TIMEOUT
  ): Long {
    return suspendCoroutine { continuation ->
      val invokeCallback: InvokeCallback =
          object : InvokeCallback {
            override fun onError(e: java.lang.Exception?) {
              continuation.resumeWithException(IllegalStateException("invoke failed", e))
            }

            override fun onResponse(invokeElement: InvokeElement?, successCode: Long) {
              continuation.resume(successCode)
            }
          }
      chipDeviceController.invoke(
          invokeCallback, devicePtr, invokeElement, timedRequestTimeoutMs, imTimeoutMs)
    }
  }
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

TE_23_02/rc2

### Platform

android

joonhaengHeo, 2023-05-23:
@sahityadubey 
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/DoorLock.adoc#41-lock-door-command
defined in the above document.
It is thought to be trying to deliver the lock-door command of DoorLock Cluster (0x101).

Looking at the document above, since the field is not an Int type, but an octstr and optional value, I think tlv should be set in the following way.

`val pinCode = "1"`
`val tlvWriter = TlvWriter()`
`tlvWriter.startStructure(AnonymousTag)`
`tlvWriter.put(ContextSpecificTag(0), pinCode)`
`tlvWriter.endStructure()`
or
`val tlvWriter = TlvWriter()`
`tlvWriter.startStructure(AnonymousTag)`
`tlvWriter.endStructure()`

joonhaengHeo, 2023-05-24:
The URL I shared is the contents of the DoorLock Cluster in the Matter Application Cluster Spec. If you cannot access that URL, please check the contents of the DoorLock Cluster in the Spec document.
"Lock Door Command"

Also, write is not a command, but related to attributes, and an error occurs when the corresponding ID is readonly.
(ID 0x00 of DoorLock Cluster is a LockState attribute, read only is correct.)

sahityadubey, 2023-05-29:
Could you please provide an example of a write command for a specific attribute? It would be helpful if you could provide an example for attribute 41.

joonhaengHeo, 2023-05-30:
@sahityadubey 
You should do something similar to the code below.

`   val clusterId = 0x101
    val attributeId = 0x29
    val tlv = TlvParseUtil.encode(true) // see detailed in TlvParseUtil.kt
    val devicePtr = ChipClient.getConnectedDevicePointer(requireContext(), addressUpdateFragment.deviceId)

    ChipClient.getDeviceController(requireContext()).write(object: WriteAttributesCallback {
      override fun onError(attributePath: ChipAttributePath?, ex: java.lang.Exception?) {
        showMessage("Write ${attribute.name} failure $ex")
        Log.e(TAG, "Write ${attribute.name} failure", ex)
      }

      override fun onResponse(attributePath: ChipAttributePath?) {
        showMessage("Write ${attribute.name} success")
      }

    }, devicePtr, listOf(AttributeWriteRequest.newInstance(addressUpdateFragment.endpointId, clusterId, attributeId, tlv, Optional.empty())), 0, 0)`

sahityadubey, 2023-05-30:
@joonhaengHeo 
Thank you it is working fine for me.

Can you please also share the list of error codes and their descriptions.
How to understand the below error codes?

```
java.lang.IllegalStateException: ../../src/lib/core/TLVReader.cpp:532
java.lang.IllegalStateException: IM Status: 135 
```

closed 2023-05-30 by sahityadubey

vivdub, 2025-02-21:
@sahityadubey 
Hi,
Need your help to do the same.
I am trying to invoke a Simple On/Off, some how it's not working.

Here is the code:

```
tlvWriter.startStructure(AnonymousTag)
        tlvWriter.endStructure()
        //tlvWriter.put(ContextSpecificTag(0), false)
        val path=AttributeWriteRequest.newInstance(endpoint.toLong(), 6L, 0L, tlvWriter.getEncoded())
        chipClient.chipDeviceController.write(object :WriteAttributesCallback{
            override fun onError(attributePath: ChipAttributePath?, e: java.lang.Exception?) {
                Timber.d("tlv error")
            }

            override fun onResponse(attributePath: ChipAttributePath?) {
                Timber.d("tlv response")
            }
        }, connectedDevicePtr, listOf(path), 0,0)

```

I don't have TLVParseUtil class in my project.

Thanks