project-chip/connectedhomeip #26716: Test Harness TC-ACE-1.5

stefano055415, 2023-05-22:
TH Version: v2.7-official-TE
Sha: 821e3f8

When I try to run for example this test TC-ACE-1.5

It breaks when TH2 tries to connect

In the logs I noticed that the TH2 tries with a different code and therefore doesn't work.

Do I have to set something?

[UI_Test_Run_2023_5_22_11_49_35.log](https://github.com/project-chip/connectedhomeip/files/11529694/UI_Test_Run_2023_5_22_11_49_35.log)

stefano055415, 2023-05-22:
@bzbarsky-apple 

My payload is MT:KKBA0OMV17EV670TF00
and the discriminator is 1863

i tried adding them as test_parameters and now the TH2 communicates with the device, but unfortunately it still doesn't work, it doesn't seem to use the correct PAA

The only other parameter I'm missing is the PakeVerifier I'm trying to configure

./spake2p gen-verifier -o - -i 1000 -s "xkmqf86IYJcYNf7+4PsyeAVjsnGu5Uwpz+XWyFpIMmQ=" -p 19628801 --out -

Index,PIN Code,Iteration Count,Salt,Verifier
0,19628801,1000,xkmqf86IYJcYNf7+4PsyeAVjsnGu5Uwpz+XWyFpIMmQ=,6hDTk1Kv+m5FBwvjulQRUBu7sDCUh/6Eb9lZMYWjmpUE3Yjk1YNX5HRbGsxoxl9Jvl39NZOG7llahxI8p9kQHaC/8YGM8aIaj75dkS+tGk7UnIij5/MBZo5suFgn0UPzEg==

stefano055415, 2023-05-22:
@bzbarsky-apple 

If I run the same test not through the browser, but from the terminal on my raspberry pi the test passes

./chip-tool tests Test_TC_ACE_1_5 --nodeId 1  --paa-trust-store-path /var/paa-root-certs/ --payload MT:KKBA0OMV17EV670TF00

stefano055415, 2023-05-22:
@bzbarsky-apple 

i also tried to remove the certificate and using the test one it works also from browser as well as from terminal

bzbarsky-apple, 2023-05-22:
> My payload is MT:KKBA0OMV17EV670TF00
> and the discriminator is 1863

bzbarsky-apple, 2023-05-22:
This all sounds like questions about the test harness, which doesn't even live in this repository.  Please file the issue in https://github.com/CHIP-Specifications/chip-certification-tool-frontend or https://github.com/CHIP-Specifications/chip-certification-tool

closed 2023-05-22 by bzbarsky-apple

stefano055415, 2023-05-23:
@bzbarsky-apple 

I think so, but I don't have access to this repository, I tried to write to CSA, but no one replies.

Couldn't you suggest how or who to ask?

Thank you

bzbarsky-apple, 2023-05-23:
@cjandhyala Please see above.

cjandhyala, 2023-05-23:
@stefano055415 There is a bug in the TC-ACE-1.5 automated script, so we requested everyone to follow the steps in the verification steps doc. Pls reach to me in the slack, if you have any more questions.

stefano055415, 2023-05-24:
@cjandhyala 

Hi, in reality the problem also occurs with other TC for example TC-CADMIN-1.16, in practice always if a TH2 is used in the test

I don't know if you are the right person to ask, but to work better and not ask for information in wrong places, I would like to know if I can be added to the repo: https://github.com/CHIP-Specifications/chip-certification-tool
or if you can suggest someone who can add me

Thank you