project-chip/connectedhomeip #26719: [BUG] Unable to commission device to custom fabric on iOS 16.5 and iOS 16.4.1

ajay-gantayet, 2023-05-22:
### Reproduction steps

1. Flash ESP32C3 with firmware from Espressif's launchpad: https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml
2. Scan following QR code using my app: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT:Y.K9042C00KA0648G00
3. Commissioning via MatterSupport succeeds, and then we try to commission device to our fabric using Matter.framework, it fails while trying to establish CASE session after sigma3 message is exchanged with the following logs:

```
I (97769) chip[SC]: Received Sigma3 msg
E (97779) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC TimeI (97839) chip[EM]: <<< [E:34962r S:0 M:260716879 (Ack:2389827)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (97859) chip[IN]: (U) Sending msg 260716879 to IP address 'UDP:[FE80::1892:39CD:6308:4C5A%st1]:62999'
E (98659) chip[CR]: mbedTLS error: ECP - The signature is not valid
E (98659) chip[DL]: Long dispatch time: 797 ms, for event type 2
I (98679) chip[EM]: <<< [E:34962r S:0 M:260716880 (Ack:2389827)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
```
4. Tried the same with all-clusters-app/esp32 and lighting-app/esp32 example from connetedhomeip repo. Seeing the same issue there as well. (commit ID: 64bf86774fda0e2a5dcd6fdafb701d5cdc509aac).

### Bug prevalence

Whenever I do this.

### GitHub hash of the SDK that was being used

64bf86774fda0e2a5dcd6fdafb701d5cdc509aac

### Platform

darwin, esp32

### Platform Version(s)

iOS 16.5, 16.4.1

bzbarsky-apple, 2023-05-22:
@ajay-gantayet Can you reproduce this reliably?  We've seen some reports of this problem but have never managed to reproduce it.

If you can reproduce reliably, it would be really great if you could help to debug this.  It would help to know the following:

1. The exact way you are using Matter.framework for commissioning into your own fabric.  Are you using the automatic certificate issuer in the framework itself, or are you issuing your own certificates?  In the latter case, are you using the framework's MTRCertificates API to create the certificates or using some other CA?
2. What exact data is received by the commissionee for the Sigma1/Sigma3 messages and the AddTrustedRootCertificate and AddNOC commands?  If you are about to enable detail logging and set `enable_im_pretty_print=true` when compiling Matter for the device, that will show you the command payloads.  To log the Sigma1/Sigma3 payloads, you would need to add some logging code to CASESession.cpp.  If you want pointers on this, or just a diff that adds the relevant code, please let me know.

I would love to get to the bottom of this....

bzbarsky-apple, 2023-05-25:
> This CSR is used to generate operational certificate which is passed in the controller initialisation method

I'm not sure what "We do not use any third party APIs to create root ca or sign the CSRs" means.  This operational certificate generation (for both the device's NOC and the controller's NOC) is done by your CA, right?  I think the most important bit I need to understand here is whether you are asking Matter.framework to generate any certificates (via the MTRCertificates APIs) or whether you are generating them yourself somehow based on the CSRs involved.

As far as the logs go, the key part is understanding _exactly_ what certificates the esp32 device received both as part of commissioning and as part of the CASE handshake (so we can figure out whether the problem is with those certificates or with the validation code on the esp32 device), and neither one is present in the attached logs.

Again, to see the certificates received during commissioning you would want to compile the esp32 image with `enable_im_pretty_print=true` and detail logging enabled (and debug-level logging enabled for the esp32 build you are using).  To see the CASE payloads you would need to add logging to CASESession.  Perhaps something like this: 
[CASE-logging.patch](https://github.com/project-chip/connectedhomeip/files/11560633/CASE-logging.patch)

ajay-gantayet, 2023-05-26:
@bzbarsky-apple , we are not using MTRCertificates APIs to generate the NOCs. The NOC is generated on the cloud using ```crypto``` APIs.

I generated new logs with the logging that you suggested. I have also enabled the following flags in the ```connectedhomeip/src/app/BUILD.gn```:
- ```chip_enable_session_resumption = true```
- ```chip_access_control_policy_logging_verbosity = 2```

PFA logs  [esp32c3_logs.log](https://github.com/project-chip/connectedhomeip/files/11576189/esp32c3_logs.log).

bzbarsky-apple, 2023-05-27:
OK, so the fundamental issue is that signature verification for Sigma3 fails.  That's at least somewhat going to help us pin it down.  That log is still missing the detail logging command payloads, but those might not matter to us for now.

That signature should be created using the operational keypair on the controller, so a few questions:

1. For this controller, how exactly is the `id<MTRKeypair>` that gets passed to `initWithIPK:...` created?   That is, what API do you use and what arguments do you pass to it?  You mentioned `kSecAttrKeyTypeECSECPrimeRandom` but not what you are doing with it.
2. What does the `publicKey` implementation of that `id<MTRKeypair>` look like?
3. Which of the `signMessageECDSA*` methods does that `id<MTRKeypair>`  implement, and what does that implementation look like?

That said, you created a CSR successfully, and that used the same keypair, so I would think the keypair is ok...

ajay-gantayet, 2023-06-02:
Hi @bzbarsky-apple ,

1. I have attached the files [MTRCSRKeys.zip](https://github.com/project-chip/connectedhomeip/files/11637701/MTRCSRKeys.zip)  for the class ```MTRCSRKeys``` used to create keypair. You can check the ```- (instancetype)initWithGroupId:(NSString *) grpId``` method to get the steps used to create the keypair. 
2. The method ```- (NSData *)publicKeyData``` is for the ```publickey``` implementation.
3. The method ```- (NSData *)signMessageECDSA_DER:(NSData *)message``` contains implementation for signing a message. 
I have reused the implementation of the ```FabrickKeys``` class from the CHIP repo for implementing the class.

For the commissioning I am using the following code:

- Initialise matter controller:
```
    func initializeMTRControllerWithUserNOC(matterFabricData: ESPNodeGroup, userNOCData: ESPIssueUserNOCResponse) {
        let storage = ESPMatterStorage.shared
        let factory = MTRDeviceControllerFactory.sharedInstance()
        let factoryParams = MTRDeviceControllerFactoryParams(storage: storage)
        if let _ = matterFabricData.fabricID?.hexToDecimal {
            do {
                try factory.start(factoryParams)
            } catch {
                return
            }
            if let grpId = group?.groupID {
                let keys = MTRCSRKeys(groupId: grpId)
                if let details = matterFabricData.fabricDetails, let rootCA = details.rootCACertificate, let rootCADerBytes = MTRCertificates.convertPEMString(toDER: rootCA), let certs = userNOCData.certificates, certs.count > 0, let noc = certs[0].userNOC, let nocDerBytes = MTRCertificates.convertPEMString(toDER: noc) {
                    let params = MTRDeviceControllerStartupParams(ipk: keys.ipk, ——> returned by the MTRCSRKeys NSData * ipk 
                        operationalKeypair: keys,   ——> MTRCSRKeys instance
                        operationalCertificate: nocDerBytes, ——> returned by cloud
                        intermediateCertificate: nil, 
                        rootCertificate: rootCADerBytes) ——-> created by cloud on fabric creation
                    params.vendorID = NSNumber(value: kTestVendorId)
                    params.operationalCertificateIssuer = self
                    params.operationalCertificateIssuerQueue = DispatchQueue.main
                    if let controller = try? factory.createController(onNewFabric: params) {
                        sController = controller
                        sController?.setDeviceControllerDelegate(self, queue: self.matterQueue)
                    } else if let controller = try? factory.createController(onExistingFabric: params) {
                        sController = controller
                        sController?.setDeviceControllerDelegate(self, queue: self.matterQueue)
                    }
                    return
                }
            }
        }
    }
```
- Start commissioning:
```
    /// Start commissioning with user NOC
    /// - Parameters:
    ///   - onboardingPayload: onboarding payload
    ///   - deviceId: device id
    func startCommissioningWithUserNOC(onboardingPayload: String, deviceId: UInt64) {
        if let controller = sController, let payload = try? MTRSetupPayload(onboardingPayload: onboardingPayload) {
            do {
                try controller.setupCommissioningSession(with: payload, newNodeID: NSNumber(value: deviceId))
                if let group = self.group, let groupName = group.groupName, let deviceName = ESPMatterEcosystemInfo.shared.getDeviceName() {
                    self.uidelegate?.showLoaderInView(message: "Commissioning \(deviceName) to \(groupName)")
                }
            }
            catch {
                print("ERROR pairAccessory: \(error.localizedDescription)")
            }
        }
    }
```
- Continue commissioning after commissioning session is established:
```
    func controller(_ : MTRDeviceController, commissioningSessionEstablishmentDone error: Error?) {
        guard let _ = error else {
            let deviceId = ESPMatterDeviceManager.shared.getCurrentDeviceId()
[... 56 lines cut ...]
    ///   - groupId: group id
    ///   - response: response
    ///   - error: error
    func nodeNOCReceived(groupId: String, response: ESPAddNodeToFabricResponse?, error: Error?) {
        let deviceId = ESPMatterDeviceManager.shared.getCurrentDeviceId()
        if let completion = completion, let response = response {
            ESPMatterFabricDetails.shared.saveAddNodeToMatterFabricDetails(groupId: groupId, deviceId: deviceId, data: response)
            if let data = ESPMatterFabricDetails.shared.getAddNodeToMatterFabricDetails(groupId: groupId, deviceId: deviceId), let certs = data.certificates, certs.count > 0, let noc = certs[0].nodeNOC, let nocData = MTRCertificates.convertPEMString(toDER: noc), let fabricData = group?.fabricDetails, let rootCACert = fabricData.rootCACertificate, let rootCertData = MTRCertificates.convertPEMString(toDER: rootCACert) {        
                if let fabData = group?.fabricDetails, let catIdAdmin = fabData.catIdAdmin, let id = "FFFFFFFD\(catIdAdmin)".hexToDecimal {
                    let params = MTROperationalCertificateChain(operationalCertificate: nocData, intermediateCertificate: nil, rootCertificate: rootCertData, adminSubject: NSNumber(value: id))
                    completion(params, nil)
                }
            }
        }
    }
}
```

[... cut ...]
static let userCSR = "-----BEGIN CERTIFICATE REQUEST-----\nMIHJMHACAQAwDjEMMAoGA1UECgwDQ1NSMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAE9rXhxx2fcFE1TM1STYISGZrY06nLqPWtb6cMVw7uBzEF8N/9XOtfbZkhEV7anHQfXsy24TT7lM326jx7weccZ6AAMAoGCCqGSM49BAMCA0kAMEYCIQCzmH03GDDQIfrKrkNljCiRLpPW9KEWkJODtpJNl/tf8wIhAOqX0nfE86+wOL5F29bILScMtlUC++3BVeRkeztNAbbI\n-----END CERTIFICATE REQUEST-----"
static let userPrivateKey = "-----BEGIN PRIVATE KEY-----\nMIGHAgEAMBMGByqGSM49AgEGCCqGSM49AwEHBG0wawIBAQQgoBujhTguJ/841zpV\n+K4WFDb0aMPH717u/Z2nC1CU9nChRANCAAT2teHHHZ9wUTVMzVJNghIZmtjTqcuo\n9a1vpwxXDu4HMQXw3/1c619tmSERXtqcdB9ezLbhNPuUzfbqPHvB5xxn\n-----END PRIVATE KEY-----\n"
```

bzbarsky-apple, 2023-06-02:
@ajay-gantayet So comparing what you have above to things I know work:

1. I'm a little surprised you are storing your _public_ key in the keychain.  What I would have assumed you do is you store the private key, and then `SecKeyCopyPublicKey` to get the public key from the private key.  I have no idea whether this is a problem in practice or not.  And I think we _are_ checking that your op key pair's public key matches the opcert, so I guess I have to assume this is OK...  Especially if that's how you are getting the public key to do your CSR with.
2. I'm confused by the fact that you generate a new IPK every time.  I don't see how that can work across restarts, but I am assuming in your commissioning flow you are not restarting, so I would not have expected this to affect commissioning.

> Note: Initially we had used hardcoded CSR and key pair and that was working but that too has stopped working with the same error.

"has stopped working" after what?

doctor-xray, 2023-07-07:
I have problem when I start up the plug from Eve with my own fabric. It has gone through successfully with MatterSupport.
What step do I need to do to control the plug from Eve via iPod with my app?
I have tried the following steps:

**MatterSupport.framework:** 
perform.request
"Matter accessory" successfully created and appears in the settings of the iPhone

**Matter.framework:**
MTRDeviceControllerFactory.sharedInstance()
MTRDeviceControllerFactoryParams(storage: MTRFactoryStorage())
mTRDeviceControllerFactory.start(mTRControllerFactoryParams)
Create FabricKeys like the example of ChipTool
MTRDeviceControllerStartupParams(
            ipk: ipk,
            fabricID: fabricID,
            nocSigner: MatterDeviceControllerService.mTRKeypair
        )
<MTRDeviceControllerStartupParams>.vendorID = NSNumber(value: Self.kTestVendorId)
<MTRDeviceController>.setupCommissioningSession(with: <MTRPayload>, newNodeID: <deviceId>)
Delegate: commissioningSessionEstablishmentDone:
---> The delegate does not come if I have previously commissioned the device with MatterSupport?

What do I have to consider?

bzbarsky-apple, 2023-07-08:
> `<MTRDeviceController>.setupCommissioningSession(with: <MTRPayload>, newNodeID: <deviceId>)`

Are you using the payload MatterSupport handed you here @doctor-xray ?

doctor-xray, 2023-07-08:
Yes I add the Payload from EVE-Plug to the request before I start the pickerView from MatterSupport.  Then I use the same Payload again with my custom Fabric in setupCommissioningSession….

doctor-xray, 2023-07-08:
> > `<MTRDeviceController>.setupCommissioningSession(with: <MTRPayload>, newNodeID: <deviceId>)`
> 
> Are you using the payload MatterSupport handed you here @doctor-xray ?

Yes I add the Payload from EVE-Plug to the request before I start the pickerView from MatterSupport. Then I use the same Payload again with my custom Fabric in setupCommissioningSession….

bzbarsky-apple, 2023-07-10:
> Then I use the same Payload again with my custom Fabric in setupCommissioningSession

This does not sound right.  What I would expect is that there is one payload for the initial commissioning via MatterSupport and a _different_ payload that you use for setupCommissioningSession.  That different payload should be that one that was passed as the second argument to your MatterAddDeviceExtensionRequestHandler's commissionDevice implementation.

doctor-xray, 2023-07-10:
> This does not sound right. What I would expect is that there is one payload for the initial commissioning via MatterSupport and a _different_ payload that you use for setupCommissioningSession. That [...]

Oh great, I see, is this payload comparable to the result of an openCommissioningWindow running in MatterSupport to create another fabric ?

doctor-xray, 2023-07-20:
> 

Ok I used than after MatterSupport Commissioning is finished the onboardingPayload from MatterAddDeviceExtensionRequestHandler  and start my own Commissioning:

<MTRDeviceController>.setupCommissioningSession(with: <MTRPayload>, newNodeID: <deviceId>)

New is thats deviceBeingCommissioned has now the UDP-SessionTransportType!!!

So I need to add in MTRCommissioningParameters the threadOperationalDataset from my iPod?
how do I get the data or is there an alternative solution?

I get currently this error:

2023-07-20 18:00:16.908226+0200 Hager Matter[9329:1798547] [Controller] Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors
2023-07-20 18:00:17.031379+0200 Hager Matter[9329:1798552] [Controller] Creating NSError from /Library/Caches/com.apple.xbs/Sources/CHIPFramework/connectedhomeip/src/darwin/Framework/CHIP/MTRDeviceAttestationDelegateBridge.mm:49: CHIP Error 0x00000013: Integrity check failed
deviceAttestationCompleted Error Domain=MTRErrorDomain Code=8 "Integrity check failed." UserInfo={NSLocalizedDescription=Integrity check failed.}

bzbarsky-apple, 2023-07-20:
> New is thats deviceBeingCommissioned has now the UDP-SessionTransportType!!!

Right, because it's on the network already, right?

> So I need to add in MTRCommissioningParameters the threadOperationalDataset from my iPod?

MatterSupport already put it on the Thread network, no?

> Failed in verifying 'Attestation Information' command received from the device: err 101

Did you set factoryParams.productAttestationAuthorityCertificates to something useful when starting up your ontorller factory?

doctor-xray, 2023-07-20:
> 
ah i see: the Certificate from MatterSupport i have to apply then too in:

DeviceCredential from MatterSupport RequestHandler:

`        /// The device's Certification Declaration as defined in the Matter specification.
        public var certificationDeclaration: Data

        /// The device's Device Attestation Certificate as defined in the Matter specification.
        public var deviceAttestationCertificate: Data

        /// The device's Product Attestation Intermediate Certificate as defined in the Matter specification.
        public var productAttestationIntermediateCertificate: Data`
        

<MTRControllerFactoryParams>.certificationDeclarationCertificates =  [certificationDeclaration]
<MTRControllerFactoryParams>.productAttestationAuthorityCertificates  =  [productAttestationIntermediateCertificate]

I then add the two certificates to the MTRControllerFactoryParams, what do I do with the third certificate: deviceAttestationCertificate ? Where do I have to put that somewhere too?

bzbarsky-apple, 2023-07-20:
You do nothing with any of those device-provided certificates.  What you _do_ need to do is decide what product attestation roots (from the DCL) you trust and tell the framework what those roots are.

doctor-xray, 2023-07-21:
> You do nothing with any of those device-provided certificates. What you _do_ need to do is decide what product attestation roots (from the DCL) you trust and tell the framework what those roots are.

These productAttestationAuthorityCertificates do I need to apply for? 

Currently I only want to control one device with its own fabric and device controller for testing.

But if I understand correctly, this is only possible with the credentials to the ThreadNetwork my iPods?  

Are there any alternatives?

bzbarsky-apple, 2023-07-21:
> These productAttestationAuthorityCertificates do I need to apply for?

No, these are just the entities that you trust to tell you devices are certified.  It's your decision as to what those entities are.

The full set of known PAA certs can be gotten from the .der files in `credentials/production/paa-root-certs` in the source tree (might be somewhat out of date; it gets synced to the DCL periodically) or directly from the DCL (see https://blog.espressif.com/matter-distributed-compliance-ledger-dcl-4013c2376e7 or https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.2.0/nrf/ug_matter_device_dcl.html or https://csa-iot.org/certification/distributed-compliance-ledger/) for some explanations, but I don't see obvious step-by-step bits to programmatically download all PAAs if that's what you want.

> Currently I only want to control one device with its own fabric and device controller for testing.

Then you only need to trust PAA that certified that specific device.

> But if I understand correctly, this is only possible with the credentials to the ThreadNetwork my iPods?
> Are there any alternatives?

I am not sure what you are asking.  If you use MatterSupport, that will give the device the Thread credentials so it can join the Thread network.  Then you can control it, after commissioning it into your own fabric.  Are you looking for a way to have it _not_ join the Thread network?

doctor-xray, 2023-07-24:
> No, these are just the entities that you trust to tell you devices are certified. It's your decision as to what those entities are.
>
> The full set of known PAA certs can be gotten from the .der files [...]

Ok Great!!  I download and the PAA Certificate to fabric: But now I got an other Error:
-> "Failed to send Solitary ack for MessageCounter" 

What I need to do?

Full Error MSG:
2023-07-24 12:39:21.243024+0200 Hager Matter[22354:2884194] [Controller] Failed to perform commissioning step 8
2023-07-24 12:39:21.359017+0200 Hager Matter[22354:2883292] [Controller] Creating NSError from ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/controller/CHIPDeviceController.cpp:1088: CHIP Error 0x000000AC: Internal error
2023-07-24 12:39:21.359030+0200 Hager Matter[22354:2884194] [ExchangeManager] Failed to send Solitary ack for MessageCounter:153340772 on exchange 29109i:../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/messaging/ExchangeContext.cpp:144: CHIP Error 0x00000002: Connection aborted
commissioningComplete with Error: Optional(Error Domain=MTRErrorDomain Code=1 "Undefined error:172." UserInfo={NSLocalizedDescription=Undefined error:172., errorCode=172})

bzbarsky-apple, 2023-07-24:
>  factoryParams.productAttestationAuthorityCertificates = <certificate.pem> as Data 

Uh... `productAttestationAuthorityCertificates` is an array of Data, no?

Also, the certs in that array should be DER, not PEM....  why is `certificate.pem` involved here?

> Failed in verifying 'Attestation Information' command received from the device: err 101

110 is kPaaNotFound. See https://github.com/project-chip/connectedhomeip/blob/4850de6020f4a05ba2b05505745583abab667114/src/credentials/attestation_verifier/DeviceAttestationVerifier.h#L35

So the `productAttestationAuthorityCertificates` you provided does not contain the PAA that signed for this device.

doctor-xray, 2023-07-24:
> Uh... `productAttestationAuthorityCertificates` is an array of Data, no?
>
> Also, the certs in that array should be DER, not PEM.... why is `certificate.pem` involved here?
>
>
> 110 is kPaaNotFound. See [...]

Clear is as array as .der data 

Maybe I loaded the wrong certificate? 
Just discovered this:
https://github.com/project-chip/connectedhomeip/issues/26142

bzbarsky-apple, 2023-07-25:
> Maybe I loaded the wrong certificate?

Very possible, yes, depending on how you picked which one to download....

doctor-xray, 2023-07-26:
> 

I tryed with the PAA -DiGICert for Eve Products:

Get an error:

2023-07-26 10:58:45.528278+0200 Hager Matter[29775:3557553] [ExchangeManager] Failed to Send CHIP MessageCounter:190187672 on exchange 44687i sendCount: 4 max retries: 4
2023-07-26 10:58:48.416557+0200 Hager Matter[29775:3557553] [SecureChannel] CASESession timed out while waiting for a response from the peer. Current state was 1
2023-07-26 10:58:48.417072+0200 Hager Matter[29775:3557553] [Controller] Session establishment failed for <0000000000000001, 1>, error: ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:304: CHIP Error 0x00000032: Timeout.  Next retry expected to get a response to Sigma1 or fail within 8 seconds
2023-07-26 10:58:59.455536+0200 Hager Matter[29775:3557415] [ExchangeManager] Failed to Send CHIP MessageCounter:190187673 on exchange 44688i sendCount: 4 max retries: 4
2023-07-26 10:59:02.903423+0200 Hager Matter[29775:3557415] [SecureChannel] CASESession timed out while waiting for a response from the peer. Current state was 1
2023-07-26 10:59:02.903906+0200 Hager Matter[29775:3557415] [Controller] Session establishment failed for <0000000000000001, 1>, error: ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:304: CHIP Error 0x00000032: Timeout.  Next retry expected to get a response to Sigma1 or fail within 43 seconds
2023-07-26 10:59:49.558571+0200 Hager Matter[29775:3557944] [ExchangeManager] Failed to Send CHIP MessageCounter:190187674 on exchange 44689i sendCount: 4 max retries: 4
2023-07-26 10:59:52.384298+0200 Hager Matter[29775:3557944] [SecureChannel] CASESession timed out while waiting for a response from the peer. Current state was 1
2023-07-26 10:59:52.384863+0200 Hager Matter[29775:3557944] [Controller] Failed to perform commissioning step 20
2023-07-26 10:59:52.385504+0200 Hager Matter[29775:3555191] [Controller] Creating NSError from ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:304: CHIP Error 0x00000032: Timeout
commissioningComplete with Error: Optional(Error Domain=MTRErrorDomain Code=9 "Transaction timed out." UserInfo={NSLocalizedDescription=Transaction timed out.})

doctor-xray, 2023-07-26:
With DiGiCert works fine. @bzbarsky-apple  Thanks for help!

MrLiuYunPing, 2023-08-22:
> payload MatterSupport handed you here @doctor-xray ?

I also faced this issue and later found out that it called the func controller (_ controller: MTRDeviceController, statusUpdate status: MTRCommissioning Status) Status is MTRCommissioning Status. failed. I don't know why?

qawitherev, 2023-08-23:
> > 
> 
> Ok I used than after MatterSupport Commissioning is finished the onboardingPayload from MatterAddDeviceExtensionRequestHandler and start my own Commissioning:

hi @doctor-xray 
how do you get the MatterAddDeviceExtensionRequestHandler to launch from Matter Extension? and call the commissionDevice?

doctor-xray, 2023-08-23:
You can add  MatterAddDeviceExtension over Xcode-> New -> Target

doctor-xray, 2023-08-23:
@qawitherev MatterAddDeviceExtensionRequestHandler (system widget ) you can only log with OS und read it over Console.App on a Mac.

` os_log(.default,"Received request to fetch rooms in home: \(String(describing: home))")
`

You can read data in this widget via AppGroup from mainApp - but reading data from MatterAddDeviceExtensionRequestHandler to mainApp over AppGroup will not work.

qawitherev, 2023-08-23:
@doctor-xray 
how do i confirm that `RequestHandler` is launched?

qawitherev, 2023-08-24:
@doctor-xray 
thanks! now i can commission my device (I guess). been trying to figure out this stuff for over a week now. 

this is how I commission using Matter framework 
`setupCommissioningSession()` -> `commissioningSessionEstablishmentDone()` controllerDelegate callback -> `commissionNode()`

bzbarsky-apple, 2023-08-28:
> Status is MTRCommissioning Status. failed. I don't know why?

@MrLiuYunPing Please file a feedback assistant ticket per the instructions in https://github.com/project-chip/connectedhomeip/issues/23566#issuecomment-1312351343 and let me know what the ticket id is.  The logs might well explain what exactly failed.

woody-apple, 2023-10-19:
Closing this for now, please re-open if you have the feedback ticket filed and noted here.

closed 2023-10-19 by woody-apple

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/

iVoGia, 2024-06-13:
Hi @doctor-xray , 
Did you completed this issue ? 
I have a problem with Matter Device when pairing , can you help me ? 
Pls take a look at : https://developer.apple.com/forums/thread/756870
Thanks.

MrLiuYunPing, 2024-08-13:
I use Matter Support to commission device to iOS Keychain and then use the payload to commission the device to my ecosystem, then I multiadmin it to the other ecosystem like Alexa. I found that if I unpair device from my ecosystem, then I use alexa to open commission window to multiadmin the device to my system, the Matter Support will not call commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) method. I check the Matter device's log, it shows  OpCreds: Failed AddNOC request (err=7e) with Operation9. The error means fabric exist. Yes, the iOS keyChain has the device already, but why the system cann't retrive the device and call   commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) method? I found Apple Home App performs well when doing the above. What should I do? Can you help me? Thanks! @woody-apple