project-chip/connectedhomeip #26721: [BUG] [TC-ACT-1.1] EventList has to be updated as per SDK implementation

Survensa, 2023-05-22:
### Reproduction steps

1. Bring up the DUT with bridge-app.
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Run the command `./chip-tool actions read event-list 1 1`

### Bug prevalence

Whenever I try to read EventList.

### GitHub hash of the SDK that was being used

bfa05804ac3c224c9e70921a857ffcac32ccc7f4 - Master

### Platform

raspi

### Platform Version(s)

RPI4 8GB RAM

### Anything else?

As per SDK implementation the mandatory Event StateChanged and ActionFailed implemented. But in EventList attribute only StateChanged is listed and ActionFailed not listed.
[image]

bzbarsky-apple, 2023-05-22:
The bridge app never dispatches the ActionFailed event, because its actions never fail.  So listing it in EventList would be pretty odd.

@CamWms @leorozendaal  what's supposed to happen in this situation, exactly?  I would have expected EventList to list "the events I will actually dispatch".  The spec says:

> This attribute SHALL be a list of the event IDs of the events supported by the cluster instance.

But what does "supported" mean here?  ActionFailed is "supported" in bridge-app in the sense that "if there were an error executing an action it would generate the event", as the cluster spec requires, but there is never an error executing an action in this application, so in practice it never generates the event.

bzbarsky-apple, 2023-05-22:
Put another way, is this a bug in the SDK, a bug in the spec, or a bug in the test plan?

leorozendaal, 2023-05-22:
@bzbarsky-apple   I would classify this as a bug of bridge-app - one could add some code to it to make it emit this event under certain condition

bzbarsky-apple, 2023-05-22:
@leorozendaal One could, but what happens with a real bridge which never fails any actions?

leorozendaal, 2023-05-23:
for me, an entry in an EventList means the event might happen under some condition.
Meaning some client may never experience it.
So I see this in reverse: if an event is NOT in the EventList, a client will never receive it from this server

cjandhyala, 2023-08-09:
since event-list is descoped Frm matter for now, closing the issue for now. Feel free to let me know if we need to reopen @Survensa

closed 2023-08-09 by cjandhyala