project-chip/connectedhomeip #26722: [BUG] [TC-BRBINFO-1.1] EventList has to be updated as per SDK implementation

Survensa, 2023-05-22:
### Reproduction steps

1. Bring up the DUT with bridge-app.
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Run the command `./chip-tool bridgeddevicebasicinformation read event-list 1 3`

### Bug prevalence

Whenever I try to read EventList.

### GitHub hash of the SDK that was being used

bfa05804ac3c224c9e70921a857ffcac32ccc7f4 - Master

### Platform

raspi

### Platform Version(s)

RPI4 8GB RAM

### Anything else?

As per SDK implementation the mandatory Event ReachableChanged is implemented. But in EventList attribute ReachableChanged is not listed. It shows 0 entries.
[image]

bzbarsky-apple, 2023-05-22:
A "real" bridge would definitely have to support ReachableChanged, so this is a bug in the emulator bridge-app: it needs to be able to pretend one of the bridged devices became unreachable.

andy31415, 2023-08-29:
Event list is still provisional for 1.2, removing cert blocker