project-chip/connectedhomeip #26743: [Platform] CHIP-TOOL - Manual Code Parsing giving errors for ShortDiscriminator() while pairing

smishra-volansys, 2023-05-23:
### Reproduction steps / Feature

Using the Chip-Tool, parse the Payload using the following command:
**chip-tool payload parse-setup-payload <QR/Manual Code>**
Once the command has been fired, we get the following output as shown in the snap, 
[image]
While we Scan the QR Payload with the prefix "MT:", the parsing works properly, and the add device starts. 

but, the parsing of Manual Code gives an error with the Short Discriminator as shown in the above snap.
It gives the error for ShortDiscriminator while pairing.

We even tried using the getShortValue() for the Pairing command in the stack, but still gives an error. 

 

### Platform

other

### Platform Version(s)

v1.0-branch

### Type

Platform validated

bzbarsky-apple, 2023-05-23:
> the parsing of Manual Code gives an error with the Short Discriminator as shown in the above snap.

There is no error shown in the screenshot.

> It gives the error for ShortDiscriminator while pairing.

Pairing with what?  Using what APIs @smishra-volansys ?

bzbarsky-apple, 2023-05-25:
@smishra-volansys Well, something parsed setup code with a short discriminator and then is asking the resulting data structure for a long discriminator value:
```
    uint16_t GetLongValue() const
    {
        VerifyOrDie(!IsShortDiscriminator());
```
which can't possibly work, because that information is just not there.

The real question is: who is calling `GetLongValue()` and why are they doing it?  What's the stack to the assert?

smishra-volansys, 2023-05-25:
1) Pairing with what?
-> We are trying to pair a Matter End device with our Gateway (acting as a Matter Controller) for which we have developed a userspace application. Hence, our Gateway acts as a Router and the device's Manual Code in entered while pairing. 

2) Using What APIs?
-> We are using the  API ManualSetupPayloadParser which we are calling from PairingCommand for the purpose of starting the commissioning process and then adding GetLongValue() in the arguments to get the discriminator after parsing. 
-> As mentioned above, instead of GetLongValue we also tried using the GetShortValue API as the logic of that is such, it checks for the short discriminator and if not then continues sending the long discriminator without error.

3) We have created a sort of wrapper, which takes all the credentials for the purpose of commissioning, and the values required when we use the chip tool. Hence to get the value of the discriminator we are calling this function.

Now, as shown in the first Snap shared above,  the Manual Code and QR Payload are of the same device but the discriminator for both are different. 

I hope I have answered all that is required. Please let me know if I missed anything or need some more details.

bzbarsky-apple, 2023-05-25:
> We are using the API ManualSetupPayloadParser

OK, so just the C++ SDK API?

> which we are calling from PairingCommand for the purpose of starting the commissioning process and then adding GetLongValue() 

You _cannot get a long discriminator_ if you started with a manual payload.  Manual payloads only have a short discriminator.

>  we also tried using the GetShortValue API 

Yes, that makes sense if you want to get a short discriminator.

The question is: what are you then doing with this discriminator?

> the Manual Code and QR Payload are of the same device but the discriminator for both are different.

The QR code has a 12-bit discriminator.  In this case `0x1ed`.  The manual code has a 4-bit discriminator, which is the first 4 bits of the full 12-bit discriminator.  In this case that's `0x1`.  All of this is correct and as-expected.

smishra-volansys, 2023-05-29:
Basically, we are using the setup pin and the discriminator by a wrapper of the Chip tool for commissioning the device.
We want that the device should get commissioned by using either of the following QR/Manual. 
And as we know that the Manual and QR both generate different discriminators, we are trying to parse this manual and QR using the same. Hence, getting short when it is manual and the long disc. when it is QR. Currently, we are able to successfully able to do that  using the QR payload but unable to do it for the Short generated by Manual Code

smishra-volansys, 2023-06-06:
We are not using the Chip tool from the terminal, instead, we are using the code-based parsing functions to fetch the data and then set the arguments to form the command.

While we used to test it with the Chip tool on the command line, 
we used:
chip-tool pairing ble-thread <NodeId> <hex thread network key> <passcode> <discriminator>

but now, we directly use the parsing APIs from the stack itself. 

We have developed a Test Mobile Application, which eventually, Scans the QR or Asks to enter the Manual Key, both of which get parsed in the Code. The QR is parsed properly and the commissioning starts, but the Manual key has a field named short discriminator which is not being parsed properly.

bzbarsky-apple, 2023-06-06:
>  we are using the code-based parsing functions to fetch the data and then set the arguments to form the command

Fine.  What exact parsing are you doing and what command are you "forming" and how?

But the real point is this: if all you have is a manual code you _cannot get a long discriminator_.  You have to work with the short one it has.

smishra-volansys, 2023-08-29:
Hi @bzbarsky-apple,  
We are able to use the Manual code for pairing as well. 
In the Enum of Pairing Mode, we were earlier using the **blePairingMode** which we replaced with **codePairingMode** in the code. Now both the QR Payload as well as Manual Code are being parsed properly for device commissioning.

closed 2023-08-29 by smishra-volansys