project-chip/connectedhomeip #26744: [Feature] If the device is powered off for 12 hours and then powered on again, it takes a long time for HomeKit to reconnect the device

tobbe2007, 2023-05-23:
### Feature description

I ran into a problem using Matter to control my air conditioner using the Apple HomeKit APP. Here's how:
1. The HomePod and air conditioner have been added to Apple HomeKit
2. The device is powered off for 12 hours and then powered on
3. After 14 minutes, the APP changed from unresponsive state to normal and controllable state

We found that after the air conditioner was restarted and connected to wifi, it was waiting for the Sigma1 message and did not receive the Sigma1 message sent by HomeKit APP. After receiving the Sigma1 message from the HomeKit APP, sessions based on the Matter protocol are resumed, and the air conditioner is connected back.

1. Is this a normal phenomenon?
2. What is the interval for evaluating the HomeKit APP to send Sigma1?

@bzbarsky-apple 

### Platform

all

### Platform Version(s)

v1.0-branch

bzbarsky-apple, 2023-05-23:
> Is this a normal phenomenon?

Yes, depending on what exact version of the OS your HomePod is running.  In 16.4 and earlier this is normal.  16.5 should be doing better.  What exact version is being used here?

> What is the interval for evaluating the HomeKit APP to send Sigma1?

In 16.4 it's retried periodically, with a backoff.  After a while the backoff gets to about "once an hour" on average.  You probably did not reach that if the device was only off for 12 hours, but 14 minutes is not at all unexpected.

In 16.5 there is some checking for new operational advertisements, but how often that happens depends on the various bits of dns-sd infrastructure involved.

Note that the device itself can reach out when it comes up and say that it's up, but it presumably does not have subscription persistence enabled.

tobbe2007, 2023-05-24:
bzbarsky, thank you very much, you help me a lot.

###  What exact version is being used here?
I think its 16.4， I'm confirming.

### Note that the device itself can reach out when it comes up and say that it's up, but it presumably does not have subscription persistence enabled.
This feature is in Mater 1.0 spec ?  
can you tell me where to check it.

tobbe2007, 2023-05-24:
bzbarsky, thank you very much, you help me a lot.

> What exact version is being used here?

I think its 16.4， I'm confirming.

> Note that the device itself can reach out when it comes up and say that it's up, but it presumably does not have subscription persistence enabled.
This feature is in Mater 1.0 spec ?
can you tell me where to check it.

bzbarsky-apple, 2023-05-25:
> This feature is in Mater 1.0 spec ?

In the sense that it's allowed by the spec, yes.  It's not in the 1.0 SDK, though.  It was implemented for the spring 2023 SDK release.

> can you tell me where to check it.

See CHIP_CONFIG_PERSIST_SUBSCRIPTIONS or the `chip_persist_subscriptions` gn option.

woody-apple, 2023-06-23:
This should be supported in the latest iOS releases, and is supported in the SDK on the device side per @bzbarsky-apple's advice above. Closing this, but please let us know if you have any issues here.

closed 2023-06-23 by woody-apple