project-chip/connectedhomeip #26746: Update Mode Select (and Derivations) XML, CodeGen, SDK per Spec Issue Updates

marktrayer, 2023-05-23:
Active in flight spec PR:
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7037

bzbarsky-apple, 2023-08-31:
@marktrayer Can you please describe what actual XML changes are needed here?

marktrayer, 2023-08-31:
So the key thing is that two attributes have been made Provisional for 1.2 across a few of the derived clusters of ModeBase.  Does the XML need any changes to reflect that? If not, then this issue can be closed.

bzbarsky-apple, 2023-08-31:
> So the key thing is that two attributes have been made Provisional for 1.2 across a few of the derived clusters of ModeBase. 

Which two?  The linked spec change has StartUpMode being removed (not provisional) and that's the only attribute I see in there...

For StartUpMode, it's being removed in the device type, not the cluster, so the cluster definition would still need to include it, perhaps, but the device type exclude it?

Or we could remove it from the cluster definition, since in practice these clusters are only used with device types that exclude the attribute.... If we're going to do that, we need to do it ASAP and this needs an owner @marktrayer

marktrayer, 2023-08-31:
So the marking of provisional happened in a subsequent group of PRs:
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7445
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7453
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7454

If removing on this basis is fine, I can probably spin up something quick.