project-chip/connectedhomeip #26777: [Feature] Is there any data file that show all available specifications for all clusters and devices.

nmtoan91, 2023-05-24:
### Feature description

...
Hello,
I'm looking for the data file that you used to generate the documents. 
The zap file in https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/all-clusters-common/all-clusters-app.zap is not enough for obtaining the Access and Conformance for each attribute. 

Thank you,

### Platform

all

### Platform Version(s)

latest

bzbarsky-apple, 2023-05-24:
@nmtoan91 There is nothing that completely expresses the "conformance" outside of the spec itself.

The XML cluster definitions will tell you the "access" and whether the conformance is M or "something else".

>  I'm looking for the data file that you used to generate the documents.

Which documents?

nmtoan91, 2023-05-24:
@bzbarsky-apple 
The documents I mentioned were: Matter-1.0-Core-Specification.pdf, Matter-1.0-Device-Library-Specification.pdf, and Matter-1.0-Application-Cluster-Specification.pdf.

Can you provide me the link to the XML cluster definitions, please? I couldn't find it.

bzbarsky-apple, 2023-05-24:
> The documents I mentioned were: Matter-1.0-Core-Specification.pdf, Matter-1.0-Device-Library-Specification.pdf, and Matter-1.0-Application-Cluster-Specification.pdf.

Those are not based on any data file.  The tables in those documents are just directly in those documents, and are the source of truth.

The XML is hand-transcribed from those tables, and in theory should match it.  It's at https://github.com/project-chip/connectedhomeip/tree/master/src/app/zap-templates/zcl/data-model/chip

closed 2023-05-24 by bzbarsky-apple