project-chip/connectedhomeip #26788: [Query] MULTI-ADMINS/FABRICS IN MATTER PROTOCOL

IcyWaterDL, 2023-05-24:
I am a newbie Matter protocol developer.

I am trying multi-admins/fabrics features. Could you please tell me how many ways to add a new admin into a node? Is the way to add without re-open the commissioning window? I have read a blog about multi-admins using edit ACLs is here: [https://grandcentrix.net/en/blog/matter_01/]

Description about my project:
****
- Device 1: Linux PC is the matter controller role (using in the first commissioning)
- Device 2: ESP32 is running lighting-app (a end-device node).
- Device 3: Another Linux PC is the matter controller (a new admin).

bzbarsky-apple, 2023-05-24:
> Is the way to add without re-open the commissioning window?

There are a few things you could mean by that question:

1. You want to add a second admin that is on the same fabric. In this case, the existing admin just needs to update the ACLs on the device to allow access for both admins, and then you are done.  This requires that all the entities involved have certiicates issued from the same root, for the same fabric id.
2. You want to add a second admin that is on a different fabric.  In this case you can do the commissioning window bit, or if the first admin happens to have access to the certificate issuing infrastructure for the second admin it can just directly arm a fail-safe over CASE, then add the new root cert, AddNOC, etc to add the new fabric to the device.  It would need to make sure when doing AddNOC to send along the node id of the new admin so the right ACLs can get set up.  The new admin would then need to `CommissioningComplete` on its new fabric.

bzbarsky-apple, 2023-05-24:
I'm going to close this, since it's not actually actionable via code changes, but please feel free to ask any follow-up questions as needed.

closed 2023-05-24 by bzbarsky-apple