project-chip/connectedhomeip #26790: [Feature] In unknown mode, the HomeKit APP is unresponsive and unavailable

tobbe2007, 2023-05-24:
### Feature description

I ran into a problem using Matter to control my air conditioner using the Apple HomeKit APP. Here's how:
1. The HomePod and air conditioner have been added to Apple HomeKit
2. If devices in HomeKit are in cooling mode, disable HomeKit.
3. Set the device mode of the remote control to dehumidification. Open HomeKit to check the device status, but no response is displayed.

Is this phenomenon normal?

@bzbarsky-apple , thank you for your help.

### Platform

all

### Platform Version(s)

v1.0-branch

tobbe2007, 2023-05-25:
Thanks for your help.
I will submit the log and other information according to your prompt.

woody-apple, 2023-06-23:
Thanks @tobbe2007, closing this as feedback assistant is the right place to follow up for Apple Home issues.

closed 2023-06-23 by woody-apple