project-chip/connectedhomeip #26791: [BUG]

tugceakkirisken, 2023-05-24:
### Reproduction steps

1. We installed Docker on Ubuntu 18.04
 2. We downloaded Docker image
 3. We run Docker
 4. https://developers.home.google.com/codelabs/matter-device-virtual#1 we completed the steps here
 5. We created a project in Google Home Developer Console
 6. We upgraded python 3.8, we upgraded gcc and g++ 9
 7. When we run ./chef.py -zbr -d rootnode_dimmablelight_bCwGYSDpoe -t linux we get G_SOURCE_FUNC error (we need to include the whole log of the error), how can we solve this?
 8. We tried Include <glib.h>, define the callback as a GSourceFunction variable and pass that callback to the function

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

non

### Platform

vscode

### Platform Version(s)

b3e4e68a

### Anything else?

I downloaded as zip on may 22 and couldnt find "GitHub hash of the SDK that was being used".

andy31415, 2023-05-24:
I don't think ubutu 18 is supported by the chip SDK. https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md asks for 22.04

andy31415, 2023-05-24:
Could you confirm what O/S are you using? You installed docker on 18.04 ... are you running a new ubuntu on there? What is your environment?

Currently chef compiles fine on the chip-build docker image I believe, however from your description I suspect you may be using a different O/S version or glib version.

andy31415, 2023-05-24:
Also since you are following https://developers.home.google.com/codelabs/matter-device-virtual#1 , the Open source SDK is probably not the right place for issues. https://developers.home.google.com/matter/support seems like a better starting point.

andy31415, 2023-05-24:
Closing: this is a non-oss sdk guide following. Not sure what the issue is (may be ubuntu version related, likely glib-version related) however support should be through google home if following google home codelabs.

closed 2023-05-24 by andy31415

tugceakkirisken, 2023-05-26:
> Could you confirm what O/S are you using? You installed docker on 18.04 ... are you running a new ubuntu on there? What is your environment?
>
> Currently chef compiles fine on the chip-build docker [...]
Hi, I'am using ubuntu 18.04. I did all the updates and followed the steps to install docker. Also glib version is GLIBC 2.27